UnchartedBull/OctoDash #446: OctoDash stops print if connected during warmup.

MatthewCroughan, 2020-02-21:
If OctoDash is left unconnected and a print is taking place, it will stop the print in progress if a user connects. 

What's interesting about this is that shows the layers in progress (Layer 0 of 117) for example.

I am unsure if this only happens if a print is heating up ready for print, but is not currently not really in progress (laying filament down). I can test this scenario later though.

UnchartedBull, 2020-02-22:
Do you mean it stops the print via OctoPrint? Does this happen on Version 1.4.0? There were some issue with the connection, which were fixed in 1.4.0.

MatthewCroughan, 2020-02-22:
I'll check again with 1.4.0 tomorrow.

MatthewCroughan, 2020-02-27:
@UnchartedBull Apologies for taking so long to get back to you on this. Indeed this still occurs in 1.4.0. Also, it occurs if a print is running.

For example. Let's say I'm running this print:

[image]

If I tap the display, the print will stop. This has happened once already when someone was curious about my display.

I started the print remotely. It does this because when you ask the Octoprint api to connect, it indeed does connect. A reconnection stops all running jobs. I would say this is the fault of the Octoprint API. This would not be acceptable for the Espurna relay API for example. If you sent a value of `1` to `espurna-name/relay/set/0` this does not turn the relay off and then on again, but that is what is happening in the case of Octoprint.

UnchartedBull, 2020-02-27:
But this is what the API documents, so it is a fault of OctoDash. The thing is, that with 1.4.0 OctoDash first checks whether a connection is already established and only sends the connect command if there is no connection. Other users reported that this is working and OctoDash isn't issuing a reconnection command if OctoPrint is connected ...

Could you please post the reply of `/api/connection` just before you tap the screen and OctoDash triggers the reconnection?

MatthewCroughan, 2020-02-27:
@UnchartedBull I see. Thanks much for explaining those details.

So, I do have the PSUControl option enabled in the OctoDash config. Does that also check for this state before issuing a power on? I'll check these states when I'm in the workshop again in 8~ hours.

UnchartedBull, 2020-02-27:
https://github.com/UnchartedBull/OctoDash/blob/master/src/app/standby/standby.component.ts#L36 that is the relevant code bit. PSUControl always get's called to turn the printer on, no matter the current connection state. But as you already said this isn't an issue, since PSUControl does not turn the printer off and on again, if it is already on. I check if PSUControl triggers a connect command as well.

UnchartedBull, 2020-03-13:
Will be tracked in #368. My fix doesn't seem to work reliably. Thanks for letting me know!

closed 2020-03-13 by UnchartedBull