UnchartedBull/OctoDash #448: OctoDash on Android

gdachs, 2020-02-22:
A cheap, or used, android tablet would make a great OctoDash display that would work with any hardware that is running Octoprint. Even small devices without HDMI output would be useable (ZeroPi, NanoPi NEO2, ...) for running OctoPrint.

I have never done something with electron. I have only made the "Tour of Heroes" with Angular. So I wouldn't be very helpful. I have read that it might be possible to get an Angular app running on Android using cordova: https://medium.com/@EliaPalme/how-to-wrap-an-angular-app-with-apache-cordova-909024a25d79

The question is now how important is Electron for OctoDash. Does OctoDash use much more than the browser engine from Electron? If not, then it should be possible to get rid of it for the Android app.

UnchartedBull, 2020-02-24:
Electron actually supports Android as a build target, so it is fairly easy for me to create an Android Build. Another user reported issues while installing though, that's why I removed it from the releases. I can build an Android App and link it here. Would be great if you could test that and report back.

Removing Electron will be hard, since it is used to store the config and get some information, like version etc ...

UnchartedBull, 2020-02-24:
https://drive.google.com/open?id=1F_0bsY353M8-yUVpd-ZH9s8m8exZ6ahy here are the APKs for arm64, armv7l and x64. Feel free to try them out and report back if they work for you 👍

gdachs, 2020-02-25:
Wow, you lightning fast! Sorry to say I am not. I am old and very busy, so don't hold your breath.
What is is the minimum Android version needed?

UnchartedBull, 2020-02-26:
No problem, just test it when you find the time. Don't get your hopes up though ... I think anything above Android 4 would work.

MatthewCroughan, 2020-02-27:
@UnchartedBull I tried these on Android 9 and 7. Both of them fail with "There was a problem while parsing the package". Would be good to get some CI builds going via Github Actions once we figure this out.

UnchartedBull, 2020-02-27:
I would also like to see Android Support. The thing is, that while electron and electron-builder do support .apk files there is little to no documentation available about that. I also don't have any access to an Android Device so testing is super hard.

If you're interested feel free to test around. I can create you a branch with the android build enabled so you can test this on your devices :)

gdachs, 2020-02-27:
Sorry for the stupid question, but I had the impression that APKs are always zip files. When I try to unpack it with the apktool, or unzip I get error messages, because it is not a zip file. The command file tells that it is a gz file. I can decompress it with gunzip but then file tells it is a tar file that tar can't unpack. Is it possible that just the packing of the APK makes problems?

MatthewCroughan, 2020-02-27:
@UnchartedBull I can set up virtual machines for this purpose. Will endeavor to do this if you set up the branch.

UnchartedBull, 2020-02-27:
@gdachs ok that sounds very interesting. I never build an Android, nor do I had an Android device in the last years. If the zip file is corrupted and can't be parse the Parsing Error would also make sense.

@MatthewCroughan I created this branch: https://github.com/UnchartedBull/OctoDash/tree/fix/android-build if you run `npm run pack` this will create the normal .deb files along with the .apk files.

Syco54645, 2020-05-04:
@UnchartedBull I can confirm that building results in an improper apk. I tried to install on multiple devices running various android versions and all reported 
```
adb: failed to install octodash-1.4.0-arm64.apk: Failure [INSTALL_PARSE_FAILED_NOT_APK: Failed to parse /data/app/vmdl1196357454.tmp/base.apk]
```

UnchartedBull, 2020-05-04:
There is a similar issue in the electron-builder repo: https://github.com/electron-userland/electron-builder/issues/4484, although there wasn't any response. Also from what I've read online Electron doesn't support Android (but rather only the .apk package format) and there is also no plan to support Android / iOS. So this would require some major rewrite to move this over to Cordova or similar. Maybe something for the future, but for now I'm closing this.

closed 2020-05-04 by UnchartedBull

sebastien247, 2022-02-01:
Did you found any solution ?

arty01238, 2023-09-09:
Has Android build been ended? I genuinely want to be able to use this? @UnchartedBull