UnchartedBull/OctoDash #462: Support Pi Zero W.

gsl12, 2020-02-29:
Hi,

first i like Octodash. Its working fine for me , when i use a Pi 3B+.

But i have trouble to use it with a Raspberry Pi Zero W. The Xorg server is crashing with fatal IO error 11. 
Is a Pi Zero W not supported or do i miss some dependencies? 

best regards

GSL

MatthewCroughan, 2020-03-01:
@gsl12 https://github.com/MatthewCroughan/octobalena Try it via my project, Octobalena. You might want to comment out the Octoprint stuff in the Docker-Compose file, if you already have Octoprint running on a separate device.

gsl12, 2020-03-01:
@MatthewCroughan Thanks, that was my plan, running only octodash on a seperate device. The Pi should have enough power to handle this. I will check your project.

UnchartedBull, 2020-03-01:
Do you want to run OctoDash on the Raspberry Pi Zero? That isn't supported, since the Raspberry Pi Zero is reaching it's limit with OctoPrint and having OctoDash on top of that will most likely mess up prints.

gsl12, 2020-03-02:
I would like to use only octodash on a pi zero w. Octoprint will run on a different pi 3b+. So i was thinking a pi zero should be enough to use only octodash.

gdachs, 2020-03-02:
That could work, but currently you have no problem with OctoDash, but with the configuration of the Xserver. So this is not an OctoDash issue.

UnchartedBull, 2020-03-02:
Yes as @gdachs mentioned this is an issue with Xorg, but even if that get's resolved OctoDash probably won't run, since it requires armv7l and won't run on armv6. I can compile you a version for armv6 if you figure out the Xorg problem.

If this works on your Pi Zero W without any problems I can think about adding the armv6 back.

gsl12, 2020-03-02:
Any advice on how to configure xserver on a pi zero to get octodash working ?

UnchartedBull, 2020-03-02:
I don't have one here, so I can only Google stuff, which wouldn't be super beneficial ...

gsl12, 2020-03-02:
I think that will be to complicate to me. I'm not familiar with xorg Linux stuff. I could find this error in the xorg log file. 
(EE) open /dev/dri/card0: No such file or directory

But I don't know what that means or how to fix it. 
I also see the crash only when octodash is loading. The output to my hyperpixel is working. I can see the cli output on it.

MatthewCroughan, 2020-03-02:
I'm going to try doing this on a pi zero now with balena. Once I figure it out I can submit some Dockerfiles for CI

gdachs, 2020-03-02:
No one of us is born with Linux knowledge.
The Xserver is referencing the wrong driver, or it is the right one, but the driver module is not loaded, installed. 
After a little bit googling U found [that](https://www.raspberrypi.org/forums/viewtopic.php?t=54121). Maybe this helps. I can't help more, because I don't use Raspberry Pis.

MatthewCroughan, 2020-03-02:
@UnchartedBull Indeed, the build does fail on armv6. How were you going to attempt to build this for armv6? If you can give me details on that then I can make something that he can just flash to an sd card and be done with, or run in [Hypriot](https://blog.hypriot.com/downloads/) or Raspbian if he figured out how to install docker on it manually.

```
[octodash]    ---> Running in 8e6e04601917
[octodash]   Cloning into 'OctoDash'...
[octodash]
[octodash]   Your branch is up to date with 'origin/master'.
[octodash]   Removing intermediate container 8e6e04601917
[octodash]    ---> 78780f5df887
[octodash]   Step 7/16 : RUN cd OctoDash && npm install && npm run ng:build
[octodash]    ---> Running in ec2902db995a
[octodash]   npm ERR! cb() never called!
[octodash]
[octodash]
[octodash]
[octodash]   npm
[octodash]   ERR!
[octodash]    This is an error with npm itself. Please report this error at:
[octodash]   npm ERR!     <https://npm.community>
[octodash]
[octodash]
[octodash]   npm ERR!
[octodash]    A complete log of this run can be found in:
[octodash]   npm ERR!
[octodash]        /root/.npm/_logs/2020-03-02T11_36_39_258Z-debug.log
[octodash]
[octodash]   Removing intermediate container ec2902db995a
[octodash]   The command '/bin/sh -c cd OctoDash && npm install && npm run ng:build' returned a non-zero code: 1
```

UnchartedBull, 2020-03-02:
Are you trying to build everything on armv6? I don't know if node.js and angular work as expected in Angular. I would run the build script on a normal x64 machine and just tell electron to also build a armv6 target. That worked in the past at least.

MatthewCroughan, 2020-03-02:
That'll be it then. My current build process actually builds the app entirely from the perspective of a running machine yes. When I run `balena push [appname]` this sents the dockerfiles in my current directory to a target-arch virtual machine, in this case armv6, so yes, it is. I'll have to see about making a Dockerfile that cross compiles this via an X86 host instead of running it all on the target arch.

closed 2020-05-08 by stale[bot]

marco123445, 2021-06-26:
Is there any way to make octodash work on Rasperry Pi Zero W now? Why is the xserver keep crashing, is there a workaround?
Thanks for your help