UnchartedBull/OctoDash #465: IP Address/Hostname of Octoprint instance should be shown

MatthewCroughan, 2020-03-01:
This will eliminate confusion when people want to access the web interface locally here at the makerspace. The value in my case would be 10.0.50.1 for the IP and matt-octoprint.local for the hostname of the octoprint instance. The problem is, these values do not match up with the values inside the config, so this has to be user programmable for vanity, as the address used in the config is literally `octoprint` since all of my configuration is making use of docker containers on the one device.

UnchartedBull, 2020-03-02:
Hmm, the problem I have with this is, that the value then would not match up with the actual config value, which could lead to a lot of confusion. From my point of view the config should reflect the actual config used by OctoDash ... Is there another reason for this, except that users might try to get the OctoPrint URL from the settings?

MatthewCroughan, 2020-03-02:
In docker, you can ping hosts with their container name, since that's how the routing works. If I execute `ping octodash' for example I'll get a response from the octodash container. So, in the OctoDash config I simply say the octoprint url is `http://octoprint:80`. This url is not accessible to other devices on the network, only the local device itself. Therefore it is not useful to display to users. All of this networking is internal and requires no router on the network. If you were to follow your idea of having the config use external addresses, OctoDash would break and be unable to communicate with the printer unless you owned a router, unless the config used 127.0.0.1, which would also be unhelpful to display.

MatthewCroughan, 2020-03-02:
Users who have no idea about any of this in the makerspace should be able to see the external  address that octoprint is accessible at in case they want to access it, that's all I'm suggesting. Currently I use the config.printer.name to display the printer hostname `matt-ender3.local`, but I'd also like to display an ip just in case their mdns isn't configured properly which can occur.

UnchartedBull, 2020-03-02:
I totally get your point that users should know the public address. Changing the url field to display something different than what is actually being used currently is no option for me since then, if someone wants to change the actual ip or hostname he/she would need to edit the config.json file. Is there maybe an option for you to put something beside the printer that tells the users the IP. This could also be something for the help screen, but I would like to display the actual config values in the config menu. I'm thinking of a private/public mode here, which would show the settings for private and a help icon with markdown for public mode ...

MatthewCroughan, 2020-03-02:
Yeah there's no real way to automatically get this, as in my circumstance the device would be running in container and would think its IP address is 172.xx.xx.xx and hostname of `octodash`.

UnchartedBull, 2020-03-09:
Then there is also no easy way for OctoDash to get the correct IP of OctoPrint ...

MatthewCroughan, 2020-03-09:
Misclicked on my phone, didn't mean to close the issue.

UnchartedBull, 2020-04-27:
I decided against showing another IP in the settings. There already is some confusion around this, so showing something different than is actually being used could create even more.

closed 2020-04-27 by UnchartedBull