UnchartedBull/OctoDash #470: Pi boots to a Black screen with White cursor (non blinking) in the top left

Joeboyc2, 2020-03-03:
**Describe the bug**
After installing Octodash via the automatic install, and several hours of troubleshooting i am no further forward.
The pi boots fine and i can see the output on the screen then i goes to a black screen with a non blinking cursor in the top left hand corner.
Im able to connect to the pi via ssh and it responds to commands.

I have plugged a keyboard and mouse in to the pi but cant get away from the black screen to see the console again.

I have tried all of the others fixes that have been mentioned - setting the display to :0, i have removed and re-installed Octodash and all the relevant X dependencies but nothing seems to help

**General Information:**
 - Hardware: Raspberry Pi 3B+ with and Elegoo TFT display [Here|https://www.amazon.co.uk/gp/product/B01MRQTMTD/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1]
 - OS Info OctoPi version: 0.17.0
 - OctoDash Version v1.4.0
 - OctoPrint Version v1.3.12

Joeboyc2, 2020-03-06:
I've managed to fix this by re-installing some bits.

This is the process i followed, for anyone having the same issues:
Install the RaspberryPi Desktop Environment:
`sudo ./scripts/install-desktop `

Then reinstall the LCD drivers:
`sudo rm -rf LCD-show 
git clone https://github.com/goodtft/LCD-show.git 
chmod -R 755 LCD-show 
cd LCD-show/
sudo ./LCD35-show`

Setup console Autologin:
`sudo raspi-config`
Option "3 Boot Options"
Option "B1 Desktop / CLI"
Option "B2 Console Autologin"
Finish and reboot

closed 2020-03-06 by Joeboyc2

UnchartedBull, 2020-03-09:
Glad you figured this out! Thanks for updating the issue with your solution!