UnchartedBull/OctoDash #479: no start   -bach: xinit: command not found

jortec, 2020-03-09:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info [e.g. Raspbian Buster Lite, please also indicate if you used OctoPi]
 - OctoDash Version [e.g. v1.0.0]
 - OctoPrint Version [e.g. v1.0.0]

**Additional context**
Add any other context about the problem here.

UnchartedBull, 2020-03-09:
I can't really help without more information. What is your setup? Did you run the automatic start script?

Also this looks like a problem with your Pi, please make sure xinit is installed and functioning.

evannadeau, 2020-03-09:
If I were to guess, I think your usual is not complete due to missing dependencies. Check out the manual install information in the wiki.

[download-and-install-octodash](https://github.com/UnchartedBull/OctoDash/wiki/Installation#download-and-install-octodash)

closed 2020-04-23 by UnchartedBull