UnchartedBull/OctoDash #481: Unable to view SD CARD files plugged into Prusa MK3S

Richardgriff, 2020-03-09:
**Describe the bug**
Frist Off i would liek to say what a wonderfull product. The UI i just amazing. However i am am having issue.

I am unable to view files that are on the SD CARD plugged into my prusa mk3s.

Launch the OCTOPRINT browser UI shows the SD CARD FILES listed. Octodash is unable to see these files.

**Expected behavior**
SD files should show in octodash as they do in octoprint

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
Rasberry PI 3B+ with Offical 7 inch Touch screen
Octoprint Version 1.4.0
Octopi version 0.17.0
Octodash version 1.4.0

[image]

UnchartedBull, 2020-03-09:
SD Card files aren‘t supported right now. Not all info is available for SD Card files, thus they are omitted. Feel free to open a PR to include these, I don‘t have the time currently. Sorry!

daddiomattyb, 2020-03-09:
I am not sure what you mean by SD card files.

Richardgriff, 2020-03-09:
Thanks for the quick response. if you can point me in the direction of the file handler I can maybe take a look. Not promising anything as my programming skills are basic but willing to help.

UnchartedBull, 2020-03-09:
src/app/files/* (should be .ts / .scss and .html file) and src/app/files.component.ts are the relevant files for this :)

PR #563 (2020-04-14): Add sdcard support
Adds more complete support for browsing/printing from the sdcard in octodash.

If there are files available from both local and sdcard, then a fake top level folder is shown to select between local and sdcard. Browsing inside those folders then continues as normal and the path starts with either '/local' or '/sdcard'.

If there are only local or sdcard files, then only the relevant folder is shown. For example if sdcard support is not enabled in octoprint, then you will just see a list of local files as usual. Only difference is the path would now show '/local' at the start  [... cut]
closed 2020-04-14 by UnchartedBull