UnchartedBull/OctoDash #483: OctoDash installed but not working

daddiomattyb, 2020-03-09:
**Describe the bug**
After installing OctoPrint, connecting to the network and installing OctoDash it shows on my LCD, and Touch works, however I get the following errors every time.
Can’t retrieve layer progress : HTTP://localhost:5000/plugin/displaylayerprogress/values : 403 forbidden
Can’t retrieve printer status: HTTP://localhost:5000/api/printer : 403 forbidden
Can’t retrieve jobs: HTTP failure response for https://localhost:5000/api/job : 403 forbidden

**To Reproduce**
Steps to reproduce the behavior:
1. Install OctoPrint
2. Connect to the network wifi.
3. Install OctoDash via the script and reboot.

**Expected behavior**
OctoDash screen should show and touch interface work and be able to work normal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware Raspberry Pi 3b+ - UCTronics 3.5 " TFT 480x320
 - OS Info Used the OctoPrint latest build
 - OctoDash Version latest available
 - OctoPrint Version 0.17.0

**Additional context**
The screen and touch interface work fine, but the errors keep showing over and over.

UnchartedBull, 2020-03-09:
Probably an invalid access token. Please double check your token.

daddiomattyb, 2020-03-09:
Not sure how to even get a token. Are we talking about the same OctoDash? I did not read anything about tokens.

UnchartedBull, 2020-03-09:
It‘s called API Key in OctoDash I think. Here is the information on how to get it: http://docs.octoprint.org/en/master/api/general.html

daddiomattyb, 2020-03-09:
Thank you. I found an API section, and it looks like it is blank. Do I copy an API from Octoprint settings under my user account and type it in there?

Sent from my Verizon, Samsung Galaxy smartphone

daddiomattyb, 2020-03-09:
Thank you. I was able to have the errors go away long enough to get a screenshot. In this picture, is this where I put the API key, and if so, do I get it from Octoprint settings?
 [Image]

UnchartedBull, 2020-03-09:
You need to copy the API Key from the Settings over to there. Should be a string with big characters and numbers.

daddiomattyb, 2020-03-11:
That did it. Thank you all for your help. Its all up and running and I am printing again.

Sent from my Verizon, Samsung Galaxy smartphone

closed 2020-03-11 by UnchartedBull

floridaservices, 2020-04-08:
Just to add since this app is new for me, the API key IS case sensitive, in case anyone else comes across this in the future.

ecarpenter2, 2020-08-17:
I am still getting this error.  I have the API correct, but it rotates between the errors mentioned above.

UnchartedBull, 2020-08-18:
If you are also getting the 403 Forbidden:

Please check that the user you created the token from is allowed to use the API. Double check that the token is correct (case-sensitve, no leading or trailing spaces).

ecarpenter2, 2020-08-18:
Thanks for your reply.  Yes the API is correct.  Where do I check and see
if the API can be used?

Everett E. Carpenter, PhD, MBA
804-723-1005

Steve021171, 2021-02-12:
Well, there is NO WAY to add a API key into an existing Octodash....
Nowhere in the settings you have the capability to enter a API key. And this f*cking message never stops showing up.... so anoying.

So, HOW can I enter the API key in a running Octodash???

UnchartedBull, 2021-02-12:
just like described in the wiki: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#inputs--settings-explained

Steve021171, 2021-02-12:
Tried, no file found, it is an empty file....

Why no menu to enter settings like that? and the message should not come up every second. One time in the beginning is enough...

Steve021171, 2021-02-12:
ok, you have to stop the service to get into that...

Steve021171, 2021-02-12:
Even that didn't worked for me... I'm done, will switch to Octoscreen.
Thanx

Steve021171, 2021-02-13:
Well, I tried Octoscreen, what works great.... but has no option to turn off screen while inactive.
Went to OctoBTT, what looks like Octodash but has no option to turn off screen either.

Went back to Octodash, still same Problem. Never accepted the API key. So I reinstalled Octoprint and OctoPi from scratch. Never get past the setup screen. 
API key was correct, but every time at the end of the setup process it shows error 403.

After hours of trying and cursing I figured out that LOCALHOST or the real IP does not work, but after I put 127.0.0.1 in it starts working.
So no clue why this makes a difference, but just i case somebody has the same problem : try 127.0.0.1 instead of localhost.

regards