UnchartedBull/OctoDash #485: Help installing

jagr200, 2020-03-10:
I can not for the life of me get this installed.

I have installed Octoprint from the Octoprint image using this link. https://www.youtube.com/watch?v=mnN4HVmjafs&t=246s

I updated drivers to get my screen functioning.

I installed Octodash by ssh into my Pi and running "wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg"

When I run that it gives me a bunch of warning about Python being out of support and to update. Then it starts the install. Everything seems to go well and it says it will be started on next boot. I run "sudo reboot now" and it reboots. Comes back up but Octodash never starts.

Any ideas?

UnchartedBull, 2020-03-13:
https://github.com/UnchartedBull/OctoDash/wiki/Installation#autologin did you do these steps?

gmarth, 2020-03-13:
I too run into some trouble. For me xorg was missing. After installing it that was fixed. But i got some bugs afterwards so i reinstalled. First Octopi (Needs to update first, the current image isnt up to date), then xorg from the manual install and lastly i installed with automatic. Now its running just fine.

PrintableProps, 2020-03-30:
> then xorg from manual install <

How do you install xorg manually? I have the same issue... 
-bash xinit: command not found

gmarth, 2020-03-30:
> > then xorg from manual install <
> 
> How do you install xorg manually? I have the same issue...
> -bash xinit: command not found

From the manual installation part of the Wiki:
```
sudo apt install xserver-xorg ratpoison x11-xserver-utils xinit libgtk-3-0
nano ~/.xinitrc
```
Add the following contents:
`#!/bin/sh

    xset s off
    xset s noblank
    xset -dpms

    ratpoison&
    octodash`
make the file executable
`sudo chmod +x .xinitrc`

Not sure if it's necessary to create and edit the xinit file. I did it anyway. Hope this helps.

UnchartedBull, 2020-03-30:
Editing the .xinitrc file is necessary, since OctoDash wouldn't start otherwise (last two lines). I added xorg to the dependencies in the script, so this should be hopefully fixed from now on. Thanks again!

closed 2020-03-30 by UnchartedBull

PrintableProps, 2020-03-30:
Helped a lot. Octodash is starting now. 

I only have one last problem now. I don't have an api key. I did not log into octoprint before installing octodash. So I guess it is not possible to get the api key now?! 

I also can't log into octopi.local via pc. Did I do something wrong? Should I be able to log into the server even before configuration of octodash? I maybe habe to reinstall the entire system. Get the api key. And then install octodash. Or am I missing something?