UnchartedBull/OctoDash #500: X-server failed "no screens found(EE)"

loadedsith, 2020-03-18:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
(To be honest, this is what I did once, I don't know for sure it reproduces)
1. Install a fresh Octopi on to an 8gb sdcard with Etcher.
2. Insert SDCard into
3. Run the automatic install with `wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg`
4. Followed the steps under "Start OctoDash Automatic on Boot" on https://github.com/UnchartedBull/OctoDash/wiki/Installation#start-octodash-automatic-on-boot
5. Reboot.
6. Pi boots to an error message "no screens found(EE)" (full log below) and a login prompt $.

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General Information:**
 - Hardware:  Raspberry Pi 3b with a 3.5in Wavepear tft LCD version B
 - OS Info: OctoPi
 - OctoDash Version latest(?)
 - OctoPrint Version  >1.4; v1.3.something

**Additional context**
I would see this in my log file and on the LCD screen:

```
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    40.084] Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
[    40.090] Current Operating System: Linux octopi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l
[    40.090] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=480 bcm2708_fb.fbheight=320 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[    40.111] Build Date: 05 June 2019  12:49:54PM
[    40.117] xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
[    40.120] Current version of pixman: 0.36.0
[    40.125] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    40.125] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    40.147] (==) Log file: "/var/log/Xorg.0.log", Time: Wed Mar 18 21:56:17 2020
[    40.153] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    40.154] (==) No Layout section.  Using the first Screen section.
[    40.154] (==) No screen section available. Using defaults.
[    40.154] (**) |-->Screen "Default Screen Section" (0)
[    40.154] (**) |   |-->Monitor "<default monitor>"
[    40.155] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    40.155] (**) |   |-->Device "Allwinner A10/A13 FBDEV"
[    40.155] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    40.156] (==) Automatically adding devices
[    40.156] (==) Automatically enabling devices
[    40.156] (==) Automatically adding GPU devices
[    40.156] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    40.156] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    40.156] 	Entry deleted from font path.
[    40.156] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    40.156] 	Entry deleted from font path.
[    40.156] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    40.156] 	Entry deleted from font path.
[    40.156] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    40.156] 	Entry deleted from font path.
[    40.156] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    40.156] 	Entry deleted from font path.
[    40.156] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    40.156] 	Entry deleted from font path.
[    40.156] (==) FontPath set to:
	/usr/share/fonts/X11/misc,
	built-ins
[    40.156] (==) ModulePath set to "/usr/lib/xorg/modules"
[    40.156] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    40.156] (II) Loader magic: 0x1fcf80
[    40.156] (II) Module ABI versions:
[    40.156] 	X.Org ANSI C Emulation: 0.4
[    40.156] 	X.Org Video Driver: 24.0
[    40.156] 	X.Org XInput driver : 24.1
[    40.156] 	X.Org Server Extension : 10.0
[    40.158] (--) using VT number 2

[    40.158] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    40.159] (II) no primary bus or device found
[    40.159] (II) LoadModule: "glx"
[    40.160] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    40.175] (II) Module glx: vendor="X.Org Foundation"
[    40.175] 	compiled for 1.20.4, module version = 1.0.0
[    40.175] 	ABI class: X.Org Server Extension, version 10.0
[    40.175] (II) LoadModule: "fbturbo"
[    40.176] (WW) Warning, couldn't open module fbturbo
[    40.176] (EE) Failed to load module "fbturbo" (module does not exist, 0)
[    40.176] (EE) No drivers available.
[    40.176] (EE)
Fatal server error:
[    40.182] (EE) no screens found(EE)
[    40.184] (EE)
Please consult the The X.Org Foundation support
	 at http://wiki.x.org
 for help.
[    40.195] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    40.201] (EE)
[    40.204] (EE) Server terminated with error (1). Closing log file.
```

loadedsith, 2020-03-18:
I was able to resolve this by running 

`sudo /home/pi/scripts/install-desktop`

And rebooting.

I said "no" to running the X environment by default during install.

closed 2020-03-18 by loadedsith

Hellsnail, 2021-01-31:
My thanks, you saved my day. after several hours trying to get the lcd to work i saw your post and solved the problem. 
https://pt.aliexpress.com/item/1005001901876699.html?spm=a2g0s.9042311.0.0.6752b90aX2fz5p
https://github.com/goodtft

kekko7072, 2021-02-05:
You saved me to. Thankyou

Outlaws2000, 2024-07-01:
> I was able to resolve this by running
> 
> `sudo /home/pi/scripts/install-desktop`
> 
> And rebooting.
> 
> I said "no" to running the X environment by default during install.

You Are amazing! was trying for 2 days, your fix did the trick!!!!