UnchartedBull/OctoDash #518: OctoDash keeps scrolling through 404 errors on the screen

J4CKT, 2020-03-25:
**What doesn't work?**
Error 404 for everything in octodash

**What did you already try?**
Followed install instructions, octodash starts automatically on every second reboot (that's another issue), I went through the startup wizard and came up with error 403 for everything.

I generated an API key in octoprint, input that key into here:
nano ~/.config/octodash/config.json
in the section "accessToken"
then did sudo reboot now

Now it gives error 404 for everything.

**General Information:**
 - Hardware [e.g. Raspberry Pi 3b+, 800x480 screen]
 - OS Info [octoprint image from their website]
 - OctoDash Version [e.g. v1.4.0]
 - OctoPrint Version [e.g. v0.17.0]

J4CKT, 2020-03-25:
My config:

[image]

UnchartedBull, 2020-03-25:
From what I can see you‘re missing the „//„ behind http:, as well as the /api/ Part.  So your URL should be http://localhost:5000/api/

UnchartedBull, 2020-03-25:
Did you create the config manually or change that somehow? DisplayLayprogress.enabled should be always true currently

J4CKT, 2020-03-25:
Thank you for the fast response. I just updated displaylayer progress and the url. This is my config now:

[image]

I have also tried using the api key from:

User settings in top right of octoprint aswell as from settings -> API. I have also tried ticking the box for Allow Cross Origin Resource Sharing. Still getting the Error 404 in bottom right of screen. The errors at the top now display:

Failed to execute "open" on xmlHTTPrequest invalid url

J4CKT, 2020-03-25:
Oh wow, i just saw the the last / after api. Just updated that and no more errors. Thank you so much.

UnchartedBull, 2020-03-25:
I think your best shot probably is to delete the file (copy the API Key first) and just go through the setup again ... This will ensure a valid config. Otherwise I'm seeing that you're missing the / behind api

UnchartedBull, 2020-03-25:
Just hit send when I saw this. Great this is working for you now!

closed 2020-03-25 by UnchartedBull