UnchartedBull/OctoDash #527: Automatic Install not working since python2.7 shutdown

PrintableProps, 2020-03-29:
Hey there, 

I was trying to install octodash via automatic installation but when I do type in the command it says:

"DEPRECATION: Python 2.7 reached the end of its life..." 

and the installation progress stops. 

Do you have any ideas to solve this? 

I updated pip already but still it does not work

UnchartedBull, 2020-03-29:
It has nothing to do with OctoDash, but rather the OctoPrint plugins that are installed during the process. I saw similar messages, although they are only warnings and won't normally stop the installation process. Python2.x is still somewhat maintained and should still work.

How long have you waited? The script runs with the quiet option, so you may need to wait a couple of minutes. Alternatively you could try to manually install the plugins yourself ...

PrintableProps, 2020-03-29:
I gave the manual installation a try but the first step which was to install the display layer progress didn't work because the master.zip wasn't available if I'm not wrong..

I will reinstall the entire system and give it a try again. 

Can you tell me if there is a solution for uploading files via USB Stick? That's my main achievement.

UnchartedBull, 2020-03-29:
Oh you probably need to install that via the OctoDash UI. Need to update the manual installation guide.

USB Upload isn‘t supported with OctoDash atm.

PrintableProps, 2020-03-30:
As you said I just had to wait while the automatic setup does its silent magic.

closed 2020-03-30 by PrintableProps

UnchartedBull, 2020-03-30:
I added an output so the script will now tell you to wait :)

PrintableProps, 2020-03-30:
Perfect. Btw I just sent you another mail. Und ich glaube übrigens du kommst auch aus Deutschland... Deshalb kannst du gerne auf deutsch antworten wenn du magst