UnchartedBull/OctoDash #536: Cannot connect to octoprint

yaawl, 2020-03-31:
Hi, 

I have a problem connecting my printer to octodash. It writes "Cant retrieve printer status" error for my url to octoprint. Does anyone have the same problem or could you please advise me? Thank you very much!

[image]
[image]
[image]

UnchartedBull, 2020-03-31:
Somehow your config seems to be really messed up. Did you edit you config manually? Please go into the settings menu and adjust your config (Don‘t include your token anywhere except in the field that das say API Tolen). Host / IP should only include the IP of your Pi with no slashes etc.

fafo17, 2020-04-11:
I've the same issue, did you solved that?

UnchartedBull, 2020-04-12:
Please provide more information in either this issue or a new one. This way I, or someone else can help you. But deleting the config for OctoDash and starting from scratch again might be your best shot for now.

closed 2020-04-27 by UnchartedBull