UnchartedBull/OctoDash #544: Development Question

jneilliii, 2020-04-05:
Sorry for the stupid question in regards to setting up a development environment on windows. I swear I've gotten this to work before but for the life of me I cannot get it to work now. Node.js is installed along with angular and angular CLI. When I attempt to start the server it seems to compile fine but when going to the webpage it just loads the background color and the logo screen for a second and then freezes up. Developer console in the browser shows this.

[image]

The way in which I ran it is opening a cmd prompt and going to the base folder where main.js file is and run `ng serve --open`.

Any advice is **greatly appreciated**.

jneilliii, 2020-04-05:
I figured out the error was from a previous version config file and after adding the missing info I was able to get it to load. The weird thing was when I modified files it would recompile in real-time and then the interface would crash out about not being able to read the config. If I changed the config and changed it back it would reload the web interface without issue. Still would like to know a little more on how to develop/debug for future changes potentially.

UnchartedBull, 2020-04-09:
So the way I normally run this is with `npm run start:big`, this will open Electron on your local machine and will open the Angular App within that Electron App (so as close to production as possible). This also uses `ng serve`, which provides the live server functionality. So as soon as any of the files of the Angular app are being changed the command will automatically recompile and reload the website. So this is completely normal behaviour :). Debugging is currently exclusively done via the Chrome (or Electron) Developer Tools. Didn't had the nerve yet to setup the remote VSCode Debugger.

If you have config errors like that, make sure to check out config.service.ts, this should tell you what OctoDash wants to have for i.e. defaultTemperatureFanSpeed. If you have any more questions let me know.

UnchartedBull, 2020-04-27:
Closing this for now. If you want more information let me know :)

closed 2020-04-27 by UnchartedBull

jeremy-3dproject, 2020-07-25:
Hello, First, thanks to  @UnchartedBull for this great application !  Thanks also for this issue that tell me how to make modifications on the source files and try it on Windows développement Tools. 

I have made some  modifications on the sources file and want to transfert them into the raspberry PI to run it directly on the lcd screen and not on the Windows environnement. Unfortunately I was a beginer in this subject and Don't know how to do this… 

The modifications I have made will be shared with you if they are working correctly. (adding a dual extruder setup, mesh bed leveling calibration wizard, nozzle cleaning feature,, etc...)

Do I need to create a .deb file to install it on the PI ? If yes, how can i do this ?

Thanks in advance and have a nice weekend !

Jérémy

UnchartedBull, 2020-07-26:
You can simply create a .deb file by running `npm run pack` (takes some time). This will then create all the .deb files (armv7l, arm64 and amd64) which can then be copied over to your Pi with scp. :)

jneilliii, 2020-07-26:
Did electron fix their build runners? When I tried to do this from windows a couple of weeks ago it kept failing on the electron side.

jeremy-3dproject, 2020-07-26:
Good afternoon 😊

Thank you for your fast reply.

I found how to create a deb file yesterday evening with your command npm run pack.

I work on windows 10 and like you tell me in your last email, it failed at the step electron.

I have installed ubuntu on another computer and create the .deb file on this one 😉

Now my modified Octodash is working on my PI.

I have about 15 printers and Octodash is actually working on a BCN3D Sigmax. Do you know this printer ? That one have a dual idex extruder system but without auto leveling feature. I create a new marlin firmware version to activate mesh bed leveling but with this modification I loose the screen created by BCN3D.

For this reason I copy your repository and create my version of Octodash.

This one is actually not finished and is still in development.

I don’t have your knowledge in programming but I understand fast and have the basics formation of the HTML, CSS and visual basic.

Actually Octodash is working with dual extruder.

I think it could be a good thing to work together on Octodash and help people around the world with your great created app 😊

I have a question for your, When I unplug and plug back my sd card in the printer, Octoprint don’t refresh the file on it. I need to go on the octoprint app and release and resync the micro sd card.

I try to send a gcode command (m22 for release and m21 for init sd card) but without success.

Is there a way that when I click on the file folder on Octodash to do this command without going on the octoprint app ? Or maybe create a button in the file folder for doing this.

If you want, I can share with you my Octodash version !

I wish you a good afternoon.

Best Regards,

Jérémy from Belgium

UnchartedBull, 2020-07-28:
You can also build just fine from WSL if you don't fancy VMs.

I would be more than happy to have a look at your code and merge a PR into master (if you like) as long as it breaks nothing. Feel free to create one, otherwise feel free to create a derivative from the main project, although I saw a few issues popping up regarding dual extrusion, so I would greatly appreciate your input here.

Regarding the SD Card issue: Should be possible, there is an API available for that  https://docs.octoprint.org/en/master/api/printer.html?highlight=sd#post--api-printer-sd. Init probably is the endpoint you're looking for. Easiest way probably would be to check whether the sd folder was opened and just sending a short request to that API first.

Feel free to share your repository and create PRs :) I promise that I'll look at them (it might just take a litte)