UnchartedBull/OctoDash #550: Change background color

Protoncek, 2020-04-07:
Hi!

I wonder if it's possible to change background color (which is at the moment some sort of grey) into pure black? It's kinda funny look, since on my 7 inch (original) LCD looks like i have brightness set waaay to high. I just wonder, i don't want to sound annoying by any mean...

BTW... while i'm writing... maybe someone posted already, but if i restart octoprint then octodash of course shows window with error (which is ok). But when it re-connects that window doesn't dissapear by itself, but i have to tap on it. Normal od bug?

Thanks!

Best regards, Pavel

UnchartedBull, 2020-04-09:
Hi,

Currently it is not possible to change the background color. You could always build yourself a custom version, https://github.com/UnchartedBull/OctoDash/blob/master/src/styles.scss#L45 that should be the relevant line. Might be something to think about in the future.

There is currently a PR open for that error (https://github.com/UnchartedBull/OctoDash/pull/540), will be fixed soon. Errors always need to be closed manually though ...

Protoncek, 2020-04-09:
Thank you for info. Like i said, it was just a wish, nothing serious. I guess that when all errors will be fixed then it will be time for beauty...

UnchartedBull, 2020-04-09:
I mean this is what it's all about. Just leave this issue open and maybe something picks it up, or I find the time to work on that :)

UnchartedBull, 2020-05-15:
This will be doable with the custom css implementation. I might just squeeze that in before the v2 release. See issue #616.

closed 2020-05-15 by UnchartedBull

Protoncek, 2020-08-04:
Hello!
Could someone point me to a direction how to create new custom css (change background color)? If i understand correct my above wish came true with version 2.0...

UnchartedBull, 2020-08-05:
Have a look at the NOX theme in the themes folder. You need to edit the contents of your custom-css file. I'm on mobile currently so the paths might not match up, but if I remember correctly there were some instructions in the NOX folder. You then just need to find the ids/classes of the divs you'd like to change the background color on.

Protoncek, 2020-08-06:
YES! Thank you!: That is look that i wanted. 
As long as i have a file to work with i will adjust it to my needs, it's just that i don't have enough knowledge to build it from scratch.