UnchartedBull/OctoDash #556: How to get thumbnail pictures

floridaservices, 2020-04-10:
**What doesn't work?**
I just installed OctoDash a few days ago and already love this little app it just works and it was not a bear to get working like OctoScreen.    On my installation a print job just has a  "qbert" style stand in pic and not a thumbnail that more closely represents the print job,  like the installation notes show.  This is a trivial issue to me, mostly just curious since I could not find the documentation.  

**What did you already try?**
I have a pretty vanilla OctoPrint install so I have wiped the card and started from scratch a few times now.  I see nothing in the config for this feature either.  I am just guessing, but would like confirmation, is it a slicer feature?  I am using PrusaSlicer.

**General Information:**
-Pi 3B
-OctoPi Raspian - believe it's Buster Lite
-Most up to date Octodash
-Most up to date OctoPrint - 1.4

**Additional context**
Add any other context or screenshots about the feature request here.

harleyg321, 2020-04-11:
To get thumbnails you'll need to use ufp (Ultimaker Format Package) instead of just gcode files. As the name would suggest I think only cura supports this format.

https://plugins.octoprint.org/plugins/UltimakerFormatPackage/

floridaservices, 2020-04-12:
Thanks, I thought it might be a slicer feature thanks for the confirmation.

closed 2020-04-12 by floridaservices