UnchartedBull/OctoDash #557: Some suggestions

SolarisI, 2020-04-11:
OctoDash is the best thing that you can use to increase your productivity and ease of use of a 3D printer. Using it for a month now, I thought of a couple of ideas for adding additional features to this interface. I don’t know if they are possible, but ideas are the first step to innovation.

1)	It will be very useful to add an additional button (on cancel, pause, adjust screen) that sends the M112 command to the printer. “Cancel” stops the print after executing last send GCode. M112 will stop the printer immediately. Useful in case of emergency. 
2)	The estimated end time on “printing” screen would be nice. I use Ender 3 screen to see on which time my print ends. Adding that function to OctoDash will eliminate the need of looking at it.
3)	Perhaps it is a good idea to change the custom actions menu to a scroll type. This will allow adding more custom actions than 6. 

After implementing those and a filament change screen, I really believe that OctoDash will become the ultimate essential tool for everyone that uses a 3D printer now and in the future.

Thanks for great work.

UnchartedBull, 2020-04-12:
Those will be some great additions. I need to postpone them to v2.x though. I wanted to release v2 for a long time now, so I‘ll finish the Filament Change and Bugs for the v2 release and will work on new enhancements after that. They definitely will be added though!

UnchartedBull, 2020-11-17:
Maybe a quick update: Your third suggestion will be released with v2.1.2 (setup needs to be done via the console though). The rest should hopefully be included before v3 :)