UnchartedBull/OctoDash #558: Better GUI usability for smaller Screens ~3.5"

ChaosBlades, 2020-04-11:
Some GUI touch elements and text are smaller than others. I suggest doing a pass on all the UI to make sure all the touch elements / text are roughly the same size within reason. The big one at the moment is the setting's menu. It is very hard to use / read on a ~3.5" touchscreen.

UnchartedBull, 2020-04-12:
Definitely would be useful, although I might not have the time right now to do that. The problem with 3.5“ is that you have very limited screen space and can only fit so much text on there. I‘ll definitely have a look at that, just don‘t expect anything too soon. I‘m always open to accept PRs though, if you want to chime in.

ChaosBlades, 2020-04-12:
Yeah, I never expected this anytime soon. Especially since this involves re-doing a lot of work. I also get the limitations of 3.5" 480x320 screens but... It looks really clean when installed on a printer. I would prefer a larger screen but it would look out of place I think.

[image]

floridaservices, 2020-04-12:
I agree that 3.5" is a good size as far as aesthetics go but small on functionality/visibility.  The screen and OctoDash's real purpose, at least for me for me, is to save me having to have another device handy to start/restart prints.  The 3.5 does this just fine with OctoDash.   Everything else it does is just icing on the cake.  I agree on the settings menu being a little small in comparision to the rest of the pages, but it's not a screen/menu that I will be in often if ever again so I didn't even think about it being an issue until i saw it here.

underpk, 2020-04-15:
I would suggest that it should have a dedicated page just for macro or custom command, it should be able to fit over 20 custom command just in one page.

Ga1j1n, 2020-05-06:
Any chance of some guidance installing a genric LCDShow 3.5 screen with Octodash, I've had a few goes hacking some bash commands from guide here and there but no success, I'm admittedly pretty amateur at command line use

here's what I've got so far, OctoPi 0.17 base install,  LCD drivers installed and functioning but no Octodash, even if I run the following first
export DISPLAY=:0
ratpoison
octodash

INSTALL SCRIPTS

sudo apt update

**Install Touchscreen driver:

Run:

git clone https://github.com/Elecrow-keen/Elecrow-LCD35.git

cd Elecrow-LCD35

sudo ./Elecrow-LCD35

reboot

cd Elecrow-LCD35

sudo ./Elecrow-LCD35 270 (If upside down, needs to install before modifying file)

reboot

***OctoDash install:

wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg

**enable Console Autologin>Boot Options**

sudo raspi-config

Upon reboot touchscreen should be up and running.

UnchartedBull, 2020-05-07:
what is showing up on the screen? If you're running OctoPi please have a look here: https://github.com/UnchartedBull/OctoDash/issues/574. Multiple people reported, that they needed to install the desktop environment to get OctoDash working (you don't need to enable autoboot to that though). Please open a new issue so this can be further discussed.

I-like-stuff, 2020-05-22:
Most 3.5” GPIO Displays also have low refresh rates, this means that for large animations e.g in the settings page or on the axis movement control buttons they behave very slowly. It would be nice, although a lot of work, to be able to disable these animations and have something else to show your touch has been registered like a change in colour.

Maybe you should add a 3.5” mode where these and other effects change place. I have no idea how you would do that but it seems to be a better way to achieve better usability as so much would have to be different.

italocjs, 2020-06-05:
Just a suggestion, you might want to take a look at astroprint touch menus for inspiration, its quite good in 3.5" screen (if you dont have the software i can send you some screenshots)

italocjs, 2020-06-06:
Another idea, for GPIO screens it might be a good idea to add a function to remove all effects, this could improve the response of the screen quite a lot.