UnchartedBull/OctoDash #559: Add connection attempt delay in X seconds for PSU Control

ChaosBlades, 2020-04-11:
For my printer you can't connect OctoPrint to the printers main board until marlin is fully booted. This is because I have an EZBoard Lite and it does not accept USB power (as it shouldn't). This means every time OctoDash comes out of sleep and turns on the printer it immediately tries and fails to connect. Then a few seconds later it connects and I have to clear a connection error.

Add option in PSU Control settings menu to delay connection attempt by X seconds.

UnchartedBull, 2020-04-12:
Which version are you using? 1.4.1 should fix exactly that, since it will try to reconnect up to 3 times (in 5s intervals) until it throws an error. If you add the 5s interval that is used if the PSUControl is used this should give the board 20s to start up.

ChaosBlades, 2020-04-12:
Yes I am using 1.4.1. This is the error I am getting.

> Can't retrieve printer status!
> Http failure response for http://localhost:5000/api/printer. 409 CONFLICT

UnchartedBull, 2020-04-14:
Ok i think there is the problem. I don't actually check against 409 Conflicts, I'll do some investigating on this and make sure to push a fix with the next version

UnchartedBull, 2020-04-14:
I did some investigating and what probably is happening is the following:

- The connection API returns a successful response with the current state being neither 'Closed' nor 'Error'
- Thus OctoDash assumes that the connection is there and exits the standby mode
- This will reactivate all services, which get the 409 error from OctoPrint.

To fix this bug I would need to response from this API: http://docs.octoprint.org/en/master/api/connection.html#get-connection-settings *WHILE* the printer is connecting (so after 1 or 2 seconds after pressing the screen).

Maybe one more thing: Have you turned on the Turn on PSU when exiting sleep option?

UnchartedBull, 2020-04-14:
Please try this dev build and report if that fixes your issue: https://drive.google.com/open?id=1eadJgAMFtbtcjNxKu33XAUU8pF8cViRw

ChaosBlades, 2020-04-20:
Just an update that I am not ghosting. My printer is torn apart right now trying to solve an issue I have been having. Thought I was going to get it resolved this weekend but that isn't going to happen as I need to order some parts. I'll test this by the end of next weekend for sure.

mikekscholz, 2020-04-23:
Try installing my fork of PortLister [https://github.com/mikekscholz/OctoPrint-PortLister](https://github.com/mikekscholz/OctoPrint-PortLister) I made it for this exact issue. It almost certainly will help you unless TH3D's firmware is radically different from mainline Marlin.

UnchartedBull, 2020-04-23:
That looks like a promising plugin. I don't have any relays setup currently, so I can't test it myself, but I make sure to link your fork in the troubleshooting guide. Thanks!

mikekscholz, 2020-04-24:
its the one I made back when I was testing the code to make tapping the standby screen automatically trigger PSU-control and they worked perfectly together.

ChaosBlades, 2020-04-24:
I was already using PortLister but not @mikekscholz fork. Looks promising. I'll be testing that and the test build of OctoDash tomorrow.

ChaosBlades, 2020-04-25:
mikekscholz's portlister fork should be considered a requirement for PSU Control enabled OctoDash setups. Such an improvement. @mikekscholz have you looked into getting this added to the OctoPrint repository? Under like OctoDash-PortLister?

As for the dev build, I am still getting a 409 error and as before I just clear the error and everything works just fine. Just to make sure I installed the build correctly I'll list the commands I used.

```
pi@octopi:/tmp $ sudo dpkg -r octodash
[sudo] password for pi:
(Reading database ... 59495 files and directories currently installed.)
Removing octodash (1.4.1) ...
Processing triggers for mime-support (3.62) ...
Processing triggers for hicolor-icon-theme (0.17-2) ...
pi@octopi:/tmp $ sudo dpkg -i octodash_1.4.1_armv7l.deb
Selecting previously unselected package octodash.
(Reading database ... 59413 files and directories currently installed.)
Preparing to unpack octodash_1.4.1_armv7l.deb ...
Unpacking octodash (1.4.1) ...
Setting up octodash (1.4.1) ...
/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.62) ...
pi@octopi:/tmp $ sudo reboot
```

UnchartedBull, 2020-04-27:
Damn ... Hoped this would fix it ... If you are able to, please provide the response from the http://docs.octoprint.org/en/master/api/connection.html#get-connection-settings while the printer is connecting (so after 1 or 2 seconds after pressing the screen). I can't really debug this, since I'm currently not using PSUControl.

Do you still get the error with mikekscholz PortLister? I already added this to the troubleshooting guide, btw 👍

ChaosBlades, 2020-04-28:
That command does not seem to work for me. I seem to be getting some kind of Adblock Plus html file as output for that. I am not running ABP on any of my systems. The output does not change regardless of connection status.

The PortLister does not change anything except I connect way faster.

UnchartedBull, 2020-04-29:
How did you query that API? With a Browser or a tool like Postman / cURL?

ChaosBlades, 2020-04-29:
I ran `GET /api/connection HTTP/1.1` through SSH via PowerShell. I love how these wikis like to make assumptions.

`pi@octopi:~ $ curl 127.0.0.1/api/connection?apikey=xxxxxxxxxx`

Is what I should have been running.

Not Connected:
`{"current":{"baudrate":null,"port":null,"printerProfile":"_default","state":"Closed"},"options":{"baudratePreference":115200,"baudrates":[250000,230400,115200,57600,38400,19200,9600],"portPreference":null,"ports":[],"printerProfilePreference":"_default","printerProfiles":[{"id":"_default","name":"Ender 3 Pro"}]}}`

Connecting:
`{"current":{"baudrate":115200,"port":"/dev/ttyACM0","printerProfile":"_default","state":"Connecting"},"options":{"baudratePreference":115200,"baudrates":[250000,230400,115200,57600,38400,19200,9600],"portPreference":null,"ports":["/dev/ttyACM0"],"printerProfilePreference":"_default","printerProfiles":[{"id":"_default","name":"Ender 3 Pro"}]}}`

Connected:
`{"current":{"baudrate":115200,"port":"/dev/ttyACM0","printerProfile":"_default","state":"Operational"},"options":{"baudratePreference":115200,"baudrates":[250000,230400,115200,57600,38400,19200,9600],"portPreference":null,"ports":["/dev/ttyACM0"],"printerProfilePreference":"_default","printerProfiles":[{"id":"_default","name":"Ender 3 Pro"}]}}`

UnchartedBull, 2020-04-29:
Ok I think I found the problem, I never changed against the state "Connecting", because it wasn't mentioned anywhere. I created another dev build: https://drive.google.com/file/d/1KL35pjbTXKDT53mZw4bDqob8yAxO2_FW/view?usp=sharing. If you have the time, check it out and report back :).

ChaosBlades, 2020-04-29:
We have a winner.
[image]

UnchartedBull, 2020-04-29:
Finally! Will be included in the next release :). You can run the dev build until then without any problems though ...

PR #569 (2020-04-29): Properly handle Connection Errors
Needs testing from issue owner.

Fixes #559.
closed 2020-04-29 by UnchartedBull