UnchartedBull/OctoDash #560: Default port number during setup should be 80 not 5000

ChaosBlades, 2020-04-11:
Unless I am mistaken the default OctoPrint port number will be 80 for 99% of installations. So that is what it should be during setup instead of 5000.

fruvous, 2020-04-11:
Port 5000 is the port that the API uses

ChaosBlades, 2020-04-11:
Weird. It would not work for me. Only port 80 worked.

nicwilson58, 2020-04-12:
5000 worked fine for me

ChaosBlades, 2020-04-12:
Does 80 work for you?

harleyg321, 2020-04-12:
The haproxy config in octopi just redirects port 80 to 127.0.0.1:5000, so in theory (assuming octodash and octoprint are running on the same pi) octodash should always work when pointed at 127.0.0.1:5000 (or localhost:5000) even if port 80 is also accessible.

Using port 80 would require haproxy set up to forward the port. Octopi may have this enabled by default (it looks like), but for example I just installed octoprint on raspbian so I don't have haproxy running.

ChaosBlades, 2020-04-12:
Ah, so my problem is that I used the same IP as when I log in from my browser instead of localhost.

Can we add this information to the Wiki Troubleshooting section?

UnchartedBull, 2020-04-12:
5000 is the default port. I‘ll add this info to the troubleshooting guide later. Thanks for the info!

UnchartedBull, 2020-04-23:
Has been added to the troubleshooting guide alongside with a brief documentation about each setting.

closed 2020-04-23 by UnchartedBull