UnchartedBull/OctoDash #562: Waveshare 3.5in Screen Not Detected

Connor205, 2020-04-12:
Hello, 
I am having the same issue as https://github.com/UnchartedBull/OctoDash/issues/199
but I am unable to fix it myself. I can see you are saying that it is the touch screen drivers that may be causing an issue, would the use of a different screen help? Just looking to get it working.
Thank you!

SolarisI, 2020-04-12:
I use Waveshare 7inch HDMI LCD (C). Installation was painless. Octopi already had the drivers build in. I only needed to set the resolution. What model of Waveshare 3.5 do you use? Turning on Raspberry Pi shows something on the screen?

floridaservices, 2020-04-12:
The elecrow 3.5 didn't work pnp like my kedei 3.5.  kedei worked as soon as I installed the driver package.  X would not acknowledge the elecrow.  I am not a Debian pro so I just stuck with the kedei.

UnchartedBull, 2020-04-12:
Please provide logs / images and the information from the template, so someone can help you. Thanks!

Connor205, 2020-04-13:
I managed to get it to work by running the install_desktop script. So I will close this issue. Thank you so much for getting back so quickly! I am using this as a replacement for octoscreen and you are amazing in comparison!

closed 2020-04-13 by Connor205