UnchartedBull/OctoDash #574: cannot open display

jaumexb, 2020-04-17:
**Describe the bug**
I have done the installation using the script but when rebooting it does not open octodash, I have tried the wiki solutions and it gives me the following error.

**Code**

pi@octopi:~ $ octodash

(octodash:825): Gtk-WARNING **: 16:00:59.360: cannot open display:
pi@octopi:~ $ ratpoison
Can't open display

floridaservices, 2020-04-17:
What kind of display are you trying to use?

jaumexb, 2020-04-17:
> What kind of display are you trying to use?

Generic 3.5" Screen

floridaservices, 2020-04-17:
What brand though, they are all different drivers and will act differently with OctoDash/X.  Ex,  A elecrow 3.5 I have would not work without additional work  but a kedei 3.5 worked without issue.

UnchartedBull, 2020-04-17:
Do you see anything in the screen? If you want to start OctoDash from the command line make sure to assign the display to your session first (export DISPLAY=:0), before running ratpoison. Or just run xinit if you‘ve set that up.

jaumexb, 2020-04-17:
> What brand though, they are all different drivers and will act differently with OctoDash/X. Ex, A elecrow 3.5 I have would not work without additional work but a kedei 3.5 worked without issue. What i'm saying is this is most likely a video driver issue with X and not a problem with OctoDash itself.

Thank you very much I will try installing the driver that you have told me, I installed the one of waveshare

jaumexb, 2020-04-17:
> Do you see anything in the screen? If you want to start OctoDash from the command line make sure to assign the display to your session first (export DISPLAY=:0), before running ratpoison. Or just run xinit if you‘ve set that up.

Yes, I can see messages from the terminal that I do not understand, I think that with verifications since "OK" appears in several green lines. I have tried the commands that you have indicated and I have obtained the result that I show below. I apologize if I do not understand much and appreciate the help.

pi@octopi:~ $ xinit

/usr/lib/xorg/Xorg.wrap: Only console users are allowed to run the X server
xinit: giving up
xinit: unable to connect to X server: Connection refused
xinit: server error
pi@octopi:~ $ export DISPLAY=:0
pi@octopi:~ $ ratpoison
Can't open display

UnchartedBull, 2020-04-23:
Could you maybe try running the `export DISPLAY=:0` before running xinit? If you have the chance you can also try entering xinit with a keyboard directly attached to the pi.

kendoodle, 2020-04-24:
I had the same problem, even after running export DISPLAY=:0 and ratpoison on the pi directly.  I would really like to have this working.

UnchartedBull, 2020-04-24:
What happens if you run xinit directly on your Pi?

philbeauchamp, 2020-04-27:
I am also getting this error.  I was able to install Octoscreen successfully so I know the screen works.  Created a second SD card from scratch to try this.  Installed Octoprint.  Updated Octoprint from inside the Web interface.  Then I did an apt-get update and upgrade.  Then I installed Octodash.  I have it set to auto-login to console.  Raspberry Pi 4 with an Osoyoo 5" touch screen.

I also get the same result as jaumexb in the above post.

UnchartedBull, 2020-04-27:
Could you maybe provide some more information regarding your setup (Raspbian Version / OctoPi / ...). I did some searching around and didn‘t find anything super useful. Maybe some things to try:

- Double Check that B2 is selected for the Boot Options (had this happen a couple times to me that I forgot to safe or so, just be sure :). Also make sure to reboot if you click Finish)
- Run xinit as Root `sudo xinit` and see if the error changes or OctoDash works
- Try reconfiguring X11: `sudo dpkg-reconfigure x11-common` and reboot to see if this fixes the problem
- Allow every User to start XOrg. Can be done via the `raspi-config` command. More information here: https://bartsimons.me/raspberry-pi-kiosk-tutorial/

Most of the information is copied from the same tutorial. If all of the above doesn‘t work maybe you could try setting up Chromium like described in the tutorial, to see whether that is working. And please report back if and what of the above worked, so I can include this in the troubleshooting guide.

@jaumexb @kendoodle did one of you managed to fix this?

philbeauchamp, 2020-04-28:
Hey!  Thanks for the response and assistance.

Octoprint version 1.4.0
OctoPi version 0.17.0

B2 was selected.

I figured out at least how to check it out.  If I choose B4 and automatically start into X, I can then start Octodash from the accessory menu.

But, with B2 selected…
_______________________________________
If I run sudo xinit, this is the outcome.

X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
Current Operating System: Linux octopi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 cma=64M cma=256M  smsc95xx.macaddr=DC:A6:32:2A:F0:C5 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=ttyS0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
Build Date: 05 June 2019  12:49:54PM
xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
Current version of pixman: 0.36.0
       Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Tue Apr 28 01:05:56 2020
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(II) modeset(0): Initializing kms color map for depth 24, 8 bpc.
xinit: Unable to run program "xterm": No such file or directory
Specify a program on the command line or make sure that /usr/bin
is in your path.

xinit: connection to X server lost

waiting for X server to shut down (II) Server terminated successfully (0). Closing log file.

Did the sudo dpkg-reconfigure x11-common.

This resulted in
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults

I could not find instructions on how to allow every user to start xOrg via that link.  It appeared to show a utility by running the dpkg-reconfigure utility.

philbeauchamp, 2020-04-28:
I was able to get around this issue by connecting a keyboard to the Pi and doing the following.

1) sudo raspi-config.  Change boot settings to go into Desktop (autologin) setting B4.  Reboot.
2) It rebooted into X server.  From there I was able to use my touchscreen and keyboard to navigate to the Preferences menu and then Default applications for LXSession.
3) In the Autostart tab, in the text box next to the "+Add", type "octodash" and then touch / click on the "+Add" button.
4) Reboot and it goes right into Octodash.

UnchartedBull, 2020-04-28:
https://unix.stackexchange.com/a/529945 here is how to manually add the anybody user to xorg. If you still want to try it, let me know if it works :)

jaumexb, 2020-04-29:
I am very happy because at last I have made it work I will put how it served me in case it serves the partner.

On the one hand I have installed everything from scratch following these steps:
1. sudo apt update && sudo apt upgrade, then run sudo apt install -f
2. install drivers
3. install octodash scrip
4. configurate autologin and reboot
5 Install deskop octoprint: sudo /home/pi/scripts/install-desktop
6. reboot and works

Thanks for your help

closed 2020-04-29 by jaumexb

kekko7072, 2021-02-05:
Hey I have the same issue but I'm not able to solve it. I installed correctly the lcd touch screen using wifi by producer (https://www.waveshare.com/wiki/5inch_HDMI_LCD) infact if I use raspberry pi desktop it works but I have issue with the x server. At the boot it shows a black screen and then the error is FATAL: Module g2d_23 not found". or unable to find screen, then  if I try to do it by ssh running octodash it says Illegal instruction. After I have installed the desktop version it not starts but the screen running raspberry desktop is working.

kekko7072, 2021-02-05:
I'am using raspberry pi3 model b and lcd directly connected using HDMI and pin. the touch screen and lcd is correctly installed.