UnchartedBull/OctoDash #579: Does OctoDash work with square aspect ratio

FricoRico, 2020-04-18:
**Additional context**
I was wondering if OctoDash would work on square aspect ratio screens such as the HyperPixel 4.0 Square. This screen is easiest to fit on my printer so I was hoping it would.

Specs to screen: https://shop.pimoroni.com/products/hyperpixel-4-square?variant=30138251477075

UnchartedBull, 2020-04-20:
I think it probably will work, it won't look good though. OctoDash is optimised for a rectangular screen layout. You can find the list of supported devices here: https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices. If you try it, make sure to let me know :)

FricoRico, 2020-04-20:
Thank you for your response! I will get the screen and check it out. If it doesn't work I'll probably patch it in through a PR.

closed 2020-04-20 by FricoRico

FricoRico, 2020-04-26:
Just reporting back, square aspect ratio works relatively well only a few alignment bugs but nothing breaking.

psk-sudo, 2020-07-13:
I was thinking about running with a square display as well. Are you able to include a screenshot with how the OctoDash looks on your Square display? Anything you did to make it work or any adjustments moving forward? 

I really appreciate it.