UnchartedBull/OctoDash #593: Can't install Octodash

Sensei1978, 2020-04-24:
I have installed all Plugins on Octoprint. Its an image. A have a raspberry pi 3 with the waveshare 5" touchscreen. But i cant install octodash. Here is my log:
pi@octopi:~ $ wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg
Installing OctoDash
Installing OctoPrint Plugins (this might take a while) ...
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Installing Dependencies ...
115 packages can be upgraded. Run 'apt list --upgradable' to see them.
libuuid1 is already the newest version (2.33.1-0.1).
libxss1 is already the newest version (1:1.2.3-1).
libxss1 set to manually installed.
libxtst6 is already the newest version (2:1.2.3-1).
libxtst6 set to manually installed.
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 gir1.2-gnomekeyring-1.0 : Depends: libgnome-keyring0 (= 3.12.0-1+b1) but it is not going to be installed
                           Depends: gir1.2-glib-2.0 but it is not going to be installed
 libappindicator3-1 : Depends: libatk1.0-0 (>= 1.12.4) but it is not going to be installed
                      Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
                      Depends: libdbusmenu-glib4 (>= 0.4.2) but it is not going to be installed
                      Depends: libdbusmenu-gtk3-4 (>= 0.4.2) but it is not going to be installed
                      Depends: libindicator3-7 but it is not going to be installed
 libgtk-3-0 : Depends: adwaita-icon-theme but it is not going to be installed
              Depends: libatk-bridge2.0-0 (>= 2.5.3) but it is not going to be installed
              Depends: libatk1.0-0 (>= 2.15.1) but it is not going to be installed
              Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
              Depends: libcolord2 (>= 0.1.10) but it is not going to be installed
              Depends: libcups2 (>= 1.6.0) but it is not going to be installed
              Depends: libepoxy0 (>= 1.4.3) but it is not going to be installed
              Depends: libjson-glib-1.0-0 (>= 0.12.0) but it is not going to be installed
              Depends: librest-0.7-0 (>= 0.7) but it is not going to be installed
              Depends: libsoup2.4-1 (>= 2.4.0) but it is not going to be installed
              Depends: libxcomposite1 (>= 1:0.3-1) but it is not going to be installed
              Depends: libgtk-3-common (>= 3.24.5-1+rpt2) but it is not going to be installed
              Recommends: libgtk-3-bin but it is not going to be installed
 libnss3 : Depends: libnspr4 (>= 2:4.12) but it is not going to be installed
 libsecret-1-0 : Depends: libsecret-common but it is not going to be installed
 octoscreen : Depends: libatk1.0-0 (>= 1.12.4) but it is not going to be installed
              Depends: libcairo-gobject2 (>= 1.10.0) but it is not going to be installed
 ratpoison : Depends: libxft2 (> 2.1.1) but it is not going to be installed
             Recommends: 9menu (>= 1.6) but it is not going to be installed
             Recommends: menu but it is not going to be installed
             Recommends: rxvt or
                         x-terminal-emulator
 x11-xserver-utils : Depends: libxaw7 but it is not going to be installed
                     Depends: libxmu6 but it is not going to be installed
                     Depends: libxt6 but it is not going to be installed
 xserver-xorg : Depends: xserver-xorg-core (>= 2:1.17.2-2) but it is not going to be installed
                Depends: xserver-xorg-video-all but it is not going to be installed or
                         xorg-driver-video
                Depends: xserver-xorg-input-all but it is not going to be installed or
                         xorg-driver-input
                Depends: x11-xkb-utils but it is not going to be installed
                Recommends: xserver-xorg-legacy but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
Installing OctoDash ...
octodash.deb                                                100%[========================================================================================================================================>]  55.35M  4.64MB/s    in 25s
(Reading database ... 47448 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (1.4.1) over (1.4.1) ...
dpkg: dependency problems prevent configuration of octodash:
 octodash depends on libgtk-3-0; however:
  Package libgtk-3-0 is not installed.
 octodash depends on libnotify4; however:
  Package libnotify4 is not installed.
 octodash depends on libnss3; however:
  Package libnss3 is not installed.
 octodash depends on xdg-utils; however:
  Package xdg-utils is not installed.
 octodash depends on libatspi2.0-0; however:
  Package libatspi2.0-0 is not installed.
 octodash depends on libappindicator3-1; however:
  Package libappindicator3-1 is not installed.
 octodash depends on libsecret-1-0; however:
  Package libsecret-1-0 is not installed.

dpkg: error processing package octodash (--install):
 dependency problems - leaving unconfigured
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.62) ...
Errors were encountered while processing:
 octodash
Setting up Autostart ...
Setting Permissions ...
chmod: cannot access '/usr/lib/xorg/Xorg': No such file or directory
Done. OctoDash will start automatically on next reboot. Please ensure that auto-login is enabled (instructions in the wiki).

UnchartedBull, 2020-04-24:
Please upgrade your Pi first: `sudo apt update && sudo apt upgrade`, then run `sudo apt install -f`. After that run the installer script again and report back.

UnchartedBull, 2020-04-24:
You also seem to have some broken dependencies on your Pi. Run `sudo apt --fix-broken install`before everything else

Sensei1978, 2020-04-24:
Thank you very much. Now its working. But octodash screen is to big for my 5" screen.

UnchartedBull, 2020-04-24:
What do you mean by it is too big?

Sensei1978, 2020-04-24:
on my 5" screen i can only see the half of the welcome screen.

UnchartedBull, 2020-04-24:
have you setup your resolution correctly? This is the first time I'm hearing of an issue like that

closed 2020-05-07 by UnchartedBull