UnchartedBull/OctoDash #595: Current height (Z) in printing status?

RC-Thoughts, 2020-04-25:
It would be nice to see the current height (Z) while printing, for example to the right of "Layer xx / yy".
It's available from current payload as "currentZ", I'm assuming printerstate is already being used so implementation should not be to hard?

UnchartedBull, 2020-04-25:
Is that available via the DLP plugin or does OctoPrint sent those values natively?

I don‘t want to clutter up the UI too much, but I‘ll have a look and see, whether this is possible to integrate.

RC-Thoughts, 2020-04-26:
It's available as push update: https://docs.octoprint.org/en/master/api/push.html?highlight=currentZ

whoim2, 2020-04-27:
Good idea, and add babystepping on job ajust screen. Customizable babystep commands, for use marlin and klipper.

UnchartedBull, 2020-04-27:
DIdn't know that the push update even existed. Did you have any experience with this? It looks super promising from the docs, so I might just replace all the GET requests with this push API in the near future (after v2 release though) ... 

@whoim2 As I said above I have to see if and how it fits in the current design. I'll have a look at that :)

RC-Thoughts, 2020-04-30:
I have not tried it, I just got my environment to compile Octodash so I can do some small things to my taste :)

UnchartedBull, 2020-04-30:
That‘s great! Let me know if you need help with anything :)

jalanjarosz, 2020-04-30:
DLP Plugin added in some new 3rd party calls.  It’s a new feature he added in recently.

Might want to see what he maid available.

Thanks!
James

RC-Thoughts, 2020-05-01:
One interesting way could be to send M114 and parse the respond with regexp like `X:([-+]?[0-9.]+) Y:([-+]?[0-9.]+) Z:([-+]?[0-9.]+) E:([-+]?[0-9.]+)` I don't think ti will be too much of time consuming to use it like every 10 sec or so. I need to play around with that a bit some day.

NickRno77, 2020-07-25:
would it be possible to disable DisplayLayerProgress Plugin from OctoDash, I've disabled app on OctoPrint and constantly get messaged telling me that DisplayLayerProgress I not installed. I've tried editing the .config/octodash/config.json and changing DLP line to false but still get error.

UnchartedBull, 2020-07-26:
Once this is merged DLP will be optional, until then it is definitely needed, since there is no way to get the Z-Height from the API. In the linked issue should be a custom build with DLP disabled though (no layer indication and fan speed).

jneilliii, 2020-12-21:
If it helps any @UnchartedBull I believe OctoFarm uses the websocket approach. Maybe something to look at as an example implementation.

UnchartedBull, 2020-12-23:
Thanks for the link. I already have a POC with the connection up and running. Will be hopefully done soonish :)

PR #1438 (2021-03-18): Switch to websockets
Not ready to merge yet. Sorry if you got all excited :/. This (currently) is just to keep track of what issues will be closed by this huge PR. If you have any thoughts - feel free to comment

Closes #1358 
Closes #1123 
Closes #595 
Closes #1177 
Closes #1374 
Closes #1505

Fixes #1304 
Fixes #1086 
Fixes #1062
Fixes #1447
Fixes #941
Fixes #1340
Fixes #1550
closed 2021-03-18 by UnchartedBull