UnchartedBull/OctoDash #596: OctoDash on my master controller can't connect to my printer

richgain, 2020-04-25:
I have installed OctoDash on my master controller but I can't connect to my printer(s).
I have three different 3D printers in my workshop. Each one has its own RPi running Octoprint.
I have a fourth Raspberry Pi with a 7" touch screen that use to control all the others.
I have installed OctoDash on this machine but I have been unable to connect to the printers.

I have managed to get OctoDash to run from the command line but when I provide the IP address and API key of a printer on the network, the config tool just reports a response failure.

Raspberry Pi 3B in printers and controller
 - OctoPi in the printer / Raspbian full desktop in the controller
 - OctoDash Version v1.4.1
 - OctoPrint Version v1.4.0]

[image]
[image]

UnchartedBull, 2020-04-25:
Regarding the startup issue: just remove the line `ratpoison&` from your `~/.xinitrc`. Otherwise Raspbian will try to create to desktops, which will not work.

Regarding the Setup error: Error Code 0 indicates that the host:Port can‘t be reached. Please make sure that your firewall allows traffic on port 5000 and that you have the proxy correctly setup for port 5000. If you run OctoPi you need to chose port 80 instead I think, if you want to access the API from another machine. So try with Port 80 first and then check your Firewall :)

richgain, 2020-04-26:
Perfect! Thank you very much.

closed 2020-04-27 by UnchartedBull