UnchartedBull/OctoDash #609: Getting errors and Octodash wont start

adminmat, 2020-05-01:
**What doesn't work?**
After what I believe was successful install (I was prompted that Octodash would launch upon reboot) I get errors and octodash will not start. 

Errors shown on Pi Screen: 
Fatal Server Error:
(EE) no screens found (EE)
Please consult the X.org Foundation support...
Server terminated with error (1) Closing log file
xinit: giving up
xinit: unable to connect to x server: Connection refused
xinit: server error

Error shown on putty terminal from my desktop when trying to lauch with "octodash":

`(octodash:833): Gtk-WARNING **: 18:10:21.469: cannot open display:`

**What did you already try?**
Updated the raspberry Pi using `sudo apt update`  and `sudo apt full-upgrade`

rebooted

installed display drivers using:
```
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```
rebooted

enabled AutoLogin to Console

rebooted

executed export DISPLAY=:0 and ratpoison before octodash

rebooted

sudo chmod ug+s /usr/lib/xorg/Xorg

rebooted

I've  tried all this several times and tried to reinstall unsing the script several times (this script: wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg)

**General Information:**
 - Hardware: Raspberry Pi 3b+ (I have been using this for Octoprint for 5 months with no issues)
 - OS Info: I've installed and am using Octoprint following these insttructions: https://octoprint.org/download/

OctoPrint version : 1.3.12
OctoPi version    : 0.17.0

**Additional context**

Screenshots of pi screen errors:
https://imgur.com/a/jMs2zHj

Screenshot of error in Putty:
https://imgur.com/MSyuNkn

floridaservices, 2020-05-01:
What brand of lcd is it, maybe you are installing drivers for a different lcd?

adminmat, 2020-05-02:
> What brand of lcd is it, maybe you are installing drivers for a different lcd?

I don't know the brand or manufacturer but it's this unit: https://www.ebay.com/itm/124138803297

https://imgur.com/a/gCx1rrP

floridaservices, 2020-05-02:
Try this repo
https://github.com/CytronTechnologies/xpt2046-LCD-Driver-for-Raspberry-Pi

adminmat, 2020-05-02:
Thanks. But this is what I explained I did in my post.

floridaservices, 2020-05-02:
> Thanks. But this is what I explained I did in my post.

You installed drivers from a different repo for a different lcd.  These need to be removed and the ones at the link I sent need to be installed.

adminmat, 2020-05-02:
Ok, I guess I don't understand the procedure.  Here is what I did. 

**1)** SSH into my pi where I run octoprint from and logged in 

**2)** enabled AutoLogin to Console 

**3)** SSH into my pi that I'm running Octoprint on and typed: wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg

**4)** This did not work so I went to the troubleshooting guide and found the section instructing you to install drivers for the display 

**5)** Installed the drivers on my pi using: 
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show

Per this page: http://www.lcdwiki.com/3.5inch_RPi_Display

(The directions from the repo you sent me says:
$ git clone https://github.com/goodtft/LCD-show.git
  $ sudo chmod -R 755 LCD-show
  $ cd LCD-show/
  $ sudo ./LCD35-show)

This did not work so I then tried all the troubleshooting procedures. 

Sorry if I am being stupid here but I'm must be missing a few steps.

floridaservices, 2020-05-02:
Your step 5 is for a different brand of lcd.  The lcd that showed in your pics had a model number which I googled and got you the right repo for that lcd.   I did not go through the installation process because I don't have this lcd brand.  You are going to need to undo step 5, then install the driver, using the repo I sent, using the instructions at that repo.  You are finding out the big catch with these cheap LCDs.  They are more often than not a huge pain in the ass to get working. You are not stupid, the process is overly complex for what it is.  Any other Linux driver is usually built into the distro so things just work, LCDs are too new a accessory for all that quite yet it would seem.

adminmat, 2020-05-02:
ok. How do i uninstall the display drivers that I installed before? 

Do I need to install Raspbian OS if I already have octoprint running? 

How do I open LX Terminal?

Thanks

floridaservices, 2020-05-02:
Someone else can probably chime in with how to uninstall.  I am not a linux pro so I found it easier to just wipe the card and reload octopi.  My setup is pretty simple so it's not a big deal to start over.

adminmat, 2020-05-02:
So you are saying that these two repos are different: 

Repo I used:           git clone https://github.com/goodtft/LCD-show.git
Repo you sent me: git clone https://github.com/goodtft/LCD-show.git

i'm so confused.

floridaservices, 2020-05-02:
Well so it does.  My apologies, that's what I get for not digging down.  Looks like you have the right driver...
so have you seen this closed issue: https://github.com/UnchartedBull/OctoDash/issues/574.  Pretty much the same issue, with a few things you can try.

UnchartedBull, 2020-05-02:
From what I can see Xorg (the tool that allows anything to be displayed on the screen (except the text console)) doesn't seem to find your screen. It would be really helpful to post your `/var/log/Xorg.0.log` as this may contain further information about what has failed exactly.

I've also seen, that you're using OctoPi, please refer to the issue @floridaservices linked above, especially the very last comment. It seems like OctoPi is doing some funky stuff with Xorg and xinit, which doesn't really comply with standard debian way of doing things.

adminmat, 2020-05-02:
> From what I can see Xorg (the tool that allows anything to be displayed on the screen (except the text console)) doesn't seem to find your screen. It would be really helpful to post your [...]

So I tried all that was mentioned in the post linked above with no success. (Maybe this part is a clue: After I install X11, it keeps telling me X11 isn't intalled. Here are some screenshots of the outputs:   https://imgur.com/a/cpuKla0

I even tried @philbeauchamp method of "Change boot settings to go into Desktop (autologin) setting B4. Reboot." But that did not help. 

Am I to understand that Octodash should not be run on a Raspberry Pi that is also running OctoPi? I didn't think I'd need a second Raspberry Pi to run octodash. Maybe that is where I'm confused here? 

Here is my log file:

`pi@v2octopi:~ $ cat /var/log/Xorg.0.log
[    40.063]
X.Org X Server 1.20.4
X Protocol Version 11, Revision 0
[    40.065] Build Operating System: Linux 4.15.0-48-generic armv8l Raspbian
[    40.067] Current Operating System: Linux v2octopi 4.19.97-v7+ #1294 SMP Thu Jan 30 13:15:58 GMT 2020 armv7l
[    40.067] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 bcm2708_fb.fbwidth=640 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc                                                    95xx.macaddr=B8:27:EB:42:29:15 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=tty1 console=ttyS0,1152                                                    00 root=/dev/mmcblk0p2 rootfstype=ext4 elevator=deadline rootwait fbcon=map:10 fbcon=font:ProFont6x11 logo.nologo
[    40.072] Build Date: 05 June 2019  12:49:54PM
[    40.074] xorg-server 2:1.20.4-1+rpt1 (https://www.debian.org/support)
[    40.075] Current version of pixman: 0.36.0
[    40.076]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    40.076] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    40.083] (==) Log file: "/var/log/Xorg.0.log", Time: Sat May  2 12:57:23 2020
[    40.084] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    40.086] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    40.087] (==) No Layout section.  Using the first Screen section.
[    40.087] (==) No screen section available. Using defaults.
[    40.087] (**) |-->Screen "Default Screen Section" (0)
[    40.087] (**) |   |-->Monitor "<default monitor>"
[    40.088] (==) No device specified for screen "Default Screen Section".
        Using the first device section listed.
[    40.088] (**) |   |-->Device "Allwinner A10/A13/A20 FBDEV"
[    40.088] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    40.088] (==) Automatically adding devices
[    40.088] (==) Automatically enabling devices
[    40.088] (==) Automatically adding GPU devices
[    40.088] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    40.088] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    40.088]    Entry deleted from font path.
[    40.088] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    40.088]    Entry deleted from font path.
[    40.088] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    40.088]    Entry deleted from font path.
[    40.088] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    40.088]    Entry deleted from font path.
[    40.089] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    40.089]    Entry deleted from font path.
[    40.089] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    40.089]    Entry deleted from font path.
[    40.089] (==) FontPath set to:
        /usr/share/fonts/X11/misc,
        built-ins
[    40.089] (==) ModulePath set to "/usr/lib/xorg/modules"
[    40.089] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    40.089] (II) Loader magic: 0x1fcf80
[    40.089] (II) Module ABI versions:
[    40.089]    X.Org ANSI C Emulation: 0.4
[    40.089]    X.Org Video Driver: 24.0
[    40.089]    X.Org XInput driver : 24.1
[    40.089]    X.Org Server Extension : 10.0
[    40.091] (--) using VT number 2

[    40.091] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    40.091] (II) no primary bus or device found
[    40.091] (II) LoadModule: "glx"
[    40.091] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    40.094] (II) Module glx: vendor="X.Org Foundation"
[    40.094]    compiled for 1.20.4, module version = 1.0.0
[    40.095]    ABI class: X.Org Server Extension, version 10.0
[    40.095] (II) LoadModule: "fbturbo"
[    40.095] (WW) Warning, couldn't open module fbturbo
[    40.095] (EE) Failed to load module "fbturbo" (module does not exist, 0)
[    40.095] (EE) No drivers available.
[    40.095] (EE)
Fatal server error:
[    40.097] (EE) no screens found(EE)
[    40.098] (EE)
Please consult the The X.Org Foundation support
         at http://wiki.x.org
 for help.
[    40.100] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[    40.102] (EE)
[    40.103] (EE) Server terminated with error (1). Closing log file.
`

adminmat, 2020-05-02:
@UnchartedBull Maybe it's because I didn't have the drivers installed when I initially installed using your script? Should I wipe the Raspberry Pi and SD card and start fresh? Having the drivers installed before running the install script?

adminmat, 2020-05-02:
Not sure if this helps but getting a fatal IO error 11

https://imgur.com/m7VQDJC

UnchartedBull, 2020-05-02:
OctoDash should run fine on OctoPi, just the setup is a little more complicated I think.

The logs are saying that you‘re missing fbturbo. Please try installing that: https://github.com/ssvb/xf86-video-fbturbo/wiki/Installation. Make sure to post the log file again once you installed that. Also the logs are saying that there are no drivers available. Maybe try reinstalling Xorg.

adminmat, 2020-05-02:
I tried that this morning. It caused the raspberry pi screen lock up when loading then I could no longer access the command line. 

I keep getting this error over and over when trying to run the screen calibrator: unable to connect to X server

What does that mean?

floridaservices, 2020-05-02:
By chance have you tried to connect a HDMI lcd to the pi during all this?  I have run into something like this before where xorg would load the app on the TV but now on the little lcd.  I would be curious if yours did the same

adminmat, 2020-05-02:
I just connected my 1080 x 1920 HDMI monitor and it shows text during boot of what looks like setup stuff. 

So I just installed Octoscreen using this tutorial: https://www.youtube.com/watch?v=OJ59hXSyBoI

I followed it exactly except installing my screen specific drivers. And it also does not start or show on my screen. So I'm going to guess here that this specific screen and the drivers do not work for this type of application. I think the best rout is to order another screen that has been proven to work and has it's own documentation and drivers. I was not trying to cheap out when I ordered this screen. I just thought it would be the safest since there are so many in use. 

Can you or someone recommend a screen that has been proven to work with Octodash and OctoPi on raspberry Pi 3B+ and 4.

Thanks

floridaservices, 2020-05-02:
I have a 3B+ and a Kedei 3.5 with OctoPi.  It is an effortless install.  I have a Elecrow 3.5 that I tried first with OctoDash and it would not work with it so I used the Kedei.

UnchartedBull, 2020-05-02:
Otherwise you could try a fresh install and go with Raspbian desktop. If that comes up you can install OctoPrint and OctoDash there then. In the linked issue there was a description on how to add OctoDash as an Auto start application.

adminmat, 2020-05-02:
> Kedei 3.5

Hmm doesn't look like I can buy a Kedi 3.5 here in the US.

floridaservices, 2020-05-03:
I have bought two of them in the past 6 months (am in US)
https://ebay.us/9ne3lh
Cannot vouch for this seller, just showing you they are here.

adminmat, 2020-05-03:
> I have bought two of them in the past 6 months (am in US)
> https://ebay.us/9ne3lh
> Cannot vouch for this seller, just showing you they are here.

Thanks!

@UnchartedBull  I did get the screen working with a fresh raspian install. How do I install octoprint on top of raspian? I've only don't it by imaging octopi on the SD card.

UnchartedBull, 2020-05-03:
https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

adminmat, 2020-05-03:
> https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspbian/2337

Thanks. I read through this today.  I decided to order another screen.

adminmat, 2020-05-03:
So I ordered this: [KeDei 3.5" LCD TFT](https://www.ebay.com/itm/KeDei-3-5-LCD-TFT-480x320-Touch-Screen-for-Raspberry-Pi-2-3-4-US-Seller-/293559339561)

and this: [OSOYOO LCD Touch Screen 3.5" HDMI ](https://www.amazon.com/OSOYOO-Screen-Monitor-Display-Raspberry/dp/B01N447AEY/ref=sr_1_40?dchild=1&keywords=raspberry+pi+3.5in+touchscreen&qid=1588461953&sr=8-40)

I'd like to try out a 5" screen. Do you think this one may work:  [OSOYOO 5 Inch DSI Touch Sceen LCD Display 800x480](https://www.amazon.com/OSOYOO-Screen-Monitor-Display-Raspberry/dp/B01N447AEY/ref=sr_1_40?dchild=1&keywords=raspberry+pi+3.5in+touchscreen&qid=1588461953&sr=8-40)

I guy on Reddit said he has used a OSOYOO screen and drivers successfully.

UnchartedBull, 2020-05-04:
I don't have experience with either of those ... But they should work I think. Maybe a good starting point is to search for a Kiosk Mode Setup with Raspbian Lite for those. If you can get Chrome running, OctoDash will also run just fine.

I'm closing this issue for now, if you encounter the same bug, please reopen.

closed 2020-05-04 by UnchartedBull

adminmat, 2020-05-14:
@floridaservices  I purchased and received the KeDei screen version 6.3. 

https://imgur.com/avdgAro

The screen is working fine: https://imgur.com/eu0iMRP

I followed this guide sent to me by the seller. I did everything exactly as he says:  http://www.epyon.be/2019/01/12/raspberry-pi-3-b-with-kedei-3-5-inch-480x320-tf-lcd-display/

After insalling OctoDash with the install script I get the same warning as the last screen:

https://imgur.com/QIr2537

I'm not over $100 into raspi screens and I still cant get OctoDash to start. Did I not install the drivers correctly? Do you have any advice on this?

adminmat, 2020-05-15:
Well good news. It started right up with I reinstalled the display driver but this time I didn't make any if the changes per the guide it shipped with. The interface is very nice. When it boots it does say something about ratpoison when it boots up. Is that normal?

floridaservices, 2020-05-15:
here are my installation notes for the Kedei 3.5 v6.3 and octodash.  These things are all I had to do to get OctoDash working with this specific screen.  I should have told you not to use their instructions, sorry about that:
LCD
-Mkdir Downloads
-cd Downloads
-wget http://en.kedei.net/raspberry/v6_1/LCD_show_v6_1_3.tar.gz(also on google drive)
tar -xvzf LCD_show_v6_1_3.tar.gz
cd LCD_show_v6_1_3
sudo ./LCD35_v

OctoDash
Install-
wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg
Update-
wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/update.sh | bash
Troubleshooting-
-Can’t retrieve status - API is wrong - it’s case sensitive
-Invalid configuration screen - delete or modify config.json @ ~/.config/octodash

adminmat, 2020-05-15:
Update: I got 

> here are my installation notes for the Kedei 3.5 v6.3 and octodash. These things are all I had to do to get OctoDash working with this specific screen. I should have told you not to use their [...]

Thanks. I do have it working now but I'll likely wipe it all and start over with your method. 

Do you know how to access a config file if I want to make more changes to the Custom Actions menu? 

And do you know if there is a way to show a preview of the file like he shows in the screenshot here:  https://raw.githubusercontent.com/TimonGaebelein/OctoprintDash/master/screenshots/file_details.png

floridaservices, 2020-05-15:
I believe that all the configuration can be done in the config.json file.  

Some more notes on my installation since you are following along - I used the Octopi standard image on a Pi 3B+.  I had a pretty vanilla setup when I set up OctoDash so it was nothing to wipe and start over - I also tried the way that people said to do it didn't work.  Then I installed only the driver without any of the tweaks and poof it worked!  And they all seem to be that way, the Elecrow and Waveshare seem to have their setup issues as well.  

The image preview is a little complicated - there's this: https://github.com/UnchartedBull/OctoDash/issues/556

And then there's a way with the PrusaSlicer plugin.  I think this is the direction that the developer is going, see https://github.com/UnchartedBull/OctoDash/issues/633.  If you happen to be using PrusaSlicer, if you install the dev version of OctoDash and configure the PrusaSlicer plugin it should work like you are wanting.

UnchartedBull, 2020-05-15:
That's great news! The ratpoison thing is completely normal, as this is the Window Manager, this should only be visible for a few seconds though.

You can also customise the Custom Actions via the UI, which actually offers the exact same fields as in the config. If you want to edit it via ssh, here you go: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#settings-file. You can't break anything really, if there are errors OctoDash will let you know :)

PrusaSlicer Preview will be included in the next release. You just need to make sure you configure either Cura or PrusaSlicer to export the thumbnail and install the correct OctoPrint Plugin (ufp or prusaslicerthumbnails).

UnchartedBull, 2020-05-15:
I also updated the troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing- to hopefully reflect what actually fixed the problem. Would be great if you could quickly glance over that and let me know if everything is there :) Thanks!

adminmat, 2020-05-15:
Thanks. I'm looking forward to the next release! So to recap: I bought a generic, waveshare clone and could not get that to work with Octodash. I then ordered the KeDei screen version 6.3 recommended by @floridaservices. This worked as long as I didn't make all the config changes that came with the instructions (as you posted in your setup troubleshooting guide). With the previous screen, I believe the drivers just aren't compatible with OctoPi or something. Not sure. However, that screen did work with Raspian desktop. 

One point I got confused while installing: When I SSH into the Pi (which was my preferred method all along) and I type "octodash" I still get a:` (octodash:950): Gtk-WARNING **: 12:02:13.802: cannot open display:` warning (on my desktop). Even thought Octodash is running and showing on the LCD. But maybe that is normal behavior anyway. 

I was also able to install Octoscreen successfully [with this screen](https://www.amazon.com/OSOYOO-Screen-Monitor-Display-Raspberry/dp/B01N447AEY/ref=sr_1_40?dchild=1&keywords=raspberry+pi+3.5in+touchscreen&qid=1588461953&sr=8-40) based on a recommendation. I have a couple notes for anyone who wants to use this same "Osoyoo" screen.

[These are the instructions](https://osoyoo.com/2017/01/18/install-3-5-hdmi-touch-screen-linux-driver-on-raspberry-pi/) referenced to install the drivers for this screen. However, if you are using a Octoprint, copying the file via USB can be time consuming. Instead I used this Wget to download and install directly from the Google Drive folder:

`wget --load-cookies /tmp/cookies.txt "https://docs.google.com/uc?export=download&confirm=$(wget --quiet --save-cookies /tmp/cookies.txt --keep-session-cookies --no-check-certificate 'https://docs.google.com/uc?export=download&id=1MVttM16SiTjvA76NPzz07Foxv4BXooyd' -O- | sed -rn 's/.*confirm=([0-9A-Za-z_]+).*/\1\n/p')&id=1MVttM16SiTjvA76NPzz07Foxv4BXooyd" -O LCD_show_35hdmi.tar.gz && rm -rf /tmp/cookies.txt`

The first time I installed the driver I used the sudo ./LCD35_480*320 This caused resolution issues. Instead, Osoyoo recommends you use a resolution of 720*480. This worked.

[Here a pic of the screen](https://imgur.com/00ZD5n7) (the Osoyoo is on the right) I'm calling it an Osoyoo although [it's branded "KeDei."](https://imgur.com/YXWIjFR)

The screen on the right is also a KeDei recommnded by @floridaservices

dennyhollick, 2020-11-13:
FYI, was having the same problem. I was able to get what I think is the same screen to work by: 

`sudo /home/pi/scripts/install-desktop
`

After configuring/calibrating the screen in the desktop, I changed my boot settings back to AutoLogin to Console and it worked fine on boot.

My screen oddly had the totally wrong calibration settings that caused the touchscreen axis to be inversed only on one axis, so I had to rejig the config files manually for it.

Jonnywayway, 2021-12-10:
Gonna add my 2 cents coming here after a long debugging session, for any future people

Most, if not all, of the TFT displays over GPIO that use the install at `https://github.com/goodtft/LCD-show.git` assume a desktop environment to install correctly and are missing the necessary checks for a desktop-less install. If you are installing something like octopi with OctoDash the safest order in which to install the files are:

1. Clean octopi 
2. Install Desktop environment `sudo /home/pi/scripts/install-desktop`
2. Install screen specific libraries [Best place to verify what library you need](http://www.lcdwiki.com/Main_Page)
3. Hard reboot
4. Verify touch works 
5. Install OctoDash `bash <(wget -qO- https://github.com/UnchartedBull/OctoDash/raw/main/scripts/install.sh)`
6. Hard reboot
7. Hopefully it works

I have also found that with some of the cheap displays, a hard reboot is required to have them work properly (no power)

slepiavka, 2022-03-15:
The summary by @Jonnywayway did it for me after hours of troubleshooting. Perhaps mentioning on the troubleshooting section? My guess is that it gets rid of any conflicting garbage individual instances might contain.