UnchartedBull/OctoDash #610: New way of resetting running values

RC-Thoughts, 2020-05-02:
Made a small change in behavior of adjusting running values from main screen, useful when playing with tuning etc.

Video here: https://youtu.be/-klEuE8iGfg

RC-Thoughts, 2020-05-02:
Examples below of the quick'n'dirty hack.

printer-status.components.ts

`private changeTemperatureHotend(value: number): void {
        this.hotendTarget += value;
        if (this.hotendTarget < -999) {
            this.hotendTarget = this.configService.getDefaultHotendTemperature();
        }
        if (this.hotendTarget < 0) {
            this.hotendTarget = 0;
        }
        if (this.hotendTarget > 999) {
            this.hotendTarget = 999;
        }
    }`

printer-status.component.html:

`<div class="quick-control__controller-value" colspan="2" *ngIf="view === QuickControlView.HOTEND" (click)="quickControlChangeValue(-1000)" matRipple
                [matRippleUnbounded]="false">
            {{ this.hotendTarget }}
            <span class="quick-control__controller-value-unit">°C</span>
        </div>`

You get the point :)

UnchartedBull, 2020-05-02:
That's a really neat idea :) Feel free to open a PR for that, is there any need to set the temp to -1000 though? I think 0 should be fine there, shouldn't it?

LanderN, 2020-05-04:
@RC-Thoughts sorry for the off-topic question, but what LCD are you using? It looks like it has very high FPS and responsiveness! Mine is a crappy cheap chinese one, which gets like 5fps, so I'm looking for a good replacement :)

RC-Thoughts, 2020-05-04:
It's Raspi 4 with Hyperpixel 4.0, really happy with it. Only downside is that no GPIO's are left over.

PR #653 (2020-05-18): Reset temperature and fan values more easy
Thanks to @RC-Thoughts for this.

Closes #610
closed 2020-05-18 by UnchartedBull