UnchartedBull/OctoDash #613: Development question

RC-Thoughts, 2020-05-03:
My goal is to do some testing on a separate computer in same network. I can do ng serve just fine, I open the localhost:4200 in browser and I see the splash-screen, but after that I get only the gray screen with source code of the splash screen.

So, apparently something is not working, is this scenario even possible? I would not like to do the dev- on the raspi connected to printer, rather on a separate comp.

My raspi with octoprint is on 192.168.1.220 and the dev-debian is on 192.168.1.8, my understanding is that this setup should work?

`root@debdev:/temp/OctoDash-master# ng serve
    WARNING: Zone.js does not support native async/await in ES2017.
    These blocks are not intercepted by zone.js and will not triggering change detection.
    See: https://github.com/angular/zone.js/pull/1140 for more information.

chunk {main} main.js, main.js.map (main) 590 kB [initial] [rendered]
chunk {polyfills} polyfills.js, polyfills.js.map (polyfills) 141 kB [initial] [rendered]
chunk {runtime} runtime.js, runtime.js.map (runtime) 6.15 kB [entry] [rendered]
chunk {styles} styles.js, styles.js.map (styles) 337 kB [initial] [rendered]
chunk {vendor} vendor.js, vendor.js.map (vendor) 6.06 MB [initial] [rendered]
Date: 2020-05-03T13:17:53.455Z - Hash: dbac93637bd39a00e5ba - Time: 24681ms
** Angular Live Development Server is listening on localhost:4200, open your browser on http://localhost:4200/ **
: Compiled successfully.`

RC-Thoughts, 2020-05-03:
There's traces of conenction in octoprint logs, IP just looks a bit combined but stoll no errors

> `2020-05-03 16:37:19,309 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.1.8
> 2020-05-03 16:37:19,654 - octoprint.server.util.flask - INFO - Passively logging in user _admin from ::ffff:192.168.1.8
> 2020-05-03 16:37:19,656 - octoprint.access.users - INFO - Logged in user: _admin
> 2020-05-03 16:37:21,799 - octoprint.server.util.sockjs - INFO - User _admin logged in on the socket from client ::ffff:192.168.1.8`

UnchartedBull, 2020-05-03:
The right command to start the dev app is `npm run start` or `npm run start:big`. This will open an electron app, which will open ng serve inside. Browser isn‘t supported, due to missing config support.

RC-Thoughts, 2020-05-03:
Works! Thank you very much!

RC-Thoughts, 2020-05-03:
Except for api-connection, need to check the octoprint-end now apparently.

UnchartedBull, 2020-05-03:
You need to enable CORS for the API in OctoPrint :)

RC-Thoughts, 2020-05-03:
We're set, thanks a lot. I simply forgot that when coming from outside the connection is ip:80 and not ip:5000 :)

Refreshing to feel like full n00b again :)

UnchartedBull, 2020-05-03:
Great that you‘re all set! Feel free to open any PRs :)

closed 2020-05-03 by UnchartedBull