UnchartedBull/OctoDash #615: only one underline in the upper left corner of the screen

twisters77, 2020-05-04:
Hello

i have installed octodash on a Raspberry Pi 4 with the latest version of octoprint according to the instructions. but after booting, only an underline appears on the screen in the upper left corner. Does anyone have an idea?

Greetings twisters

UnchartedBull, 2020-05-05:
We would need more information, like what script did you execute? What OS are you running? Did you read the Troubleshooting Guide?

Please respect the issue templates, as they are needed in order to help you.

Xenon-2510, 2020-05-06:
I had this same issue and found the Rpi was outputting the video from the HDMI, plug it into a monitor and see if you can see Octodash. If you can, then it is your screen configuration.

UnchartedBull, 2020-05-07:
If this is still an issue: please check the updated troubleshooting guide: https://github.com/UnchartedBull/OctoDash/wiki/Troubleshooting#cannot-open-display-xinit-failing-. Otherwise close this issue please.

UnchartedBull, 2020-05-11:
This is most likely the same issue as #623. If you still have this issue, please continue the conversation there. If you can't see OctoDash via HDMI please reopen this issue.

closed 2020-05-11 by UnchartedBull