UnchartedBull/OctoDash #616: Feature Request: Allow custom CSS

noxhirsch, 2020-05-04:
**Is your feature request related to a problem? Please describe.**
I'd like to adjust the style of OctoDash to my needs. At the moment this is only possible by building it myself

**Describe the solution you'd like**
Allow the user to have a theme.css file in a specific folder which would allow "endless" styling possibilities with the "official" build app (and especially without building after every style change)

**Link other projects**
If you want OctoDash to include other OctoPrint plugins or similar please link them here.

**Additional context**
Add any other context or screenshots about the feature request here.

UnchartedBull, 2020-05-05:
Makes sense. Will work on this after v2 release.

UnchartedBull, 2020-05-15:
Good news: I decided to squeeze this in before the 2.0 release. So this will be available very soon! :)

noxhirsch, 2020-05-15:
You're amazing! Thank you very much! 🎉

Mearman, 2020-05-15:
Looking forward to it

PR #656 (2020-05-18): Custom CSS
Closes #616
closed 2020-05-18 by UnchartedBull