UnchartedBull/OctoDash #620: Can't Install Octodash

dgbcoleman, 2020-05-05:
Hi there,
I am trying to install octodash and it keeps saying that I need to install the newest version of python even though I have.  Here is my command prompt:

pi@octopi:~ $ wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg
Installing OctoDash
Installing OctoPrint Plugins (this might take a while) ...
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
Installing Dependencies ...
All packages are up to date.
libappindicator3-1 is already the newest version (0.4.92-7).
libatspi2.0-0 is already the newest version (2.30.0-7).
libnotify4 is already the newest version (0.7.7-4).
libnss3 is already the newest version (2:3.42.1-1+deb10u2).
libsecret-1-0 is already the newest version (0.18.7-1).
libuuid1 is already the newest version (2.33.1-0.1).
libxss1 is already the newest version (1:1.2.3-1).
libxtst6 is already the newest version (2:1.2.3-1).
ratpoison is already the newest version (1.4.9-1).
xinit is already the newest version (1.4.0-1).
xserver-xorg is already the newest version (1:7.7+19+b8).
gir1.2-gnomekeyring-1.0 is already the newest version (3.12.0-1+b1).
libgtk-3-0 is already the newest version (3.24.5-1+rpt2).
x11-xserver-utils is already the newest version (7.7+8+rpi1).
xdg-utils is already the newest version (1.1.3-1+rpt1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Installing OctoDash ...
octodash.deb                                                100%[========================================================================================================================================>]  55.35M  80.8KB/s    in 8m 59s
(Reading database ... 56890 files and directories currently installed.)
Preparing to unpack octodash.deb ...
Unpacking octodash (1.4.1) over (1.4.1) ...
Setting up octodash (1.4.1) ...
/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.62) ...
Setting up Autostart ...
Setting Permissions ...
Done. OctoDash will start automatically on next reboot. Please ensure that auto-login is enabled (instructions in the wiki).

dgbcoleman, 2020-05-05:
I tried installing it manually but when I booted the pi it said no screens found.  I have installed my touchscreen driver already so I don't know what the problem is.

dgbcoleman, 2020-05-06:
I for got to say that I am using this lcd: https://www.waveshare.com/wiki/5inch_HDMI_LCD

dgbcoleman, 2020-05-06:
Ok, I was able to get it to work by installing the desktop environment for octoprint and saying no when it asked to auto start on startup.  Octodash is working but the bootup time is really slow, is this normal?

UnchartedBull, 2020-05-07:
Great that you got this working. What do you mean by long boot time? On my Pi 3B+ it takes approx. 10-15 seconds after Raspbian booted for everything to be ready.

dgbcoleman, 2020-05-07:
By a long time I mean that from the time that I plug in the pi to the time it starts displaying the main screen of octodash, it is about a full 60 seconds.

UnchartedBull, 2020-05-07:
60 seconds is long, but not unusually long. Mine takes 30-40 seconds, but I do have a fast SD card. If you have a slow SD Card 60 seconds might just be normal.

Closing this issue since the installation worked. If you have any more questions let me know.

closed 2020-05-07 by UnchartedBull

bsplit, 2020-08-01:
I ran this command before installing the .deb package and it was fixed
sudo apt install desktop-file-utils