UnchartedBull/OctoDash #622: Display network status icon with a current IP address

rafiki270, 2020-05-07:
With multiple printers moving between locations I struggle figuring out what IP each printer has and if it's connected at all

A network icon indicating connection on the dashboard would be amazing with a modal info screen with IP or potentially other network details available on tap

UnchartedBull, 2020-05-11:
I think this will also be related to the multi printer support, right? And you would like to see the current network status for each of the saved printers?

rafiki270, 2020-05-11:
> I think this will also be related to the multi printer support, right? And you would like to see the current network status for each of the saved printers?

I am not sure I understand the question. I thought every RaspberryPi has it's own IP and settings, not each printer?

UnchartedBull, 2020-05-11:
So you want to see the public IP of the OctoPrint instance, that OctoDash is currently connected to?

rafiki270, 2020-05-11:
> So you want to see the public IP of the OctoPrint instance, that OctoDash is currently connected to?

Yes :)

italocjs, 2020-06-06:
This is quite helpful if there are multiple printers and IP is not static

UnchartedBull, 2020-11-02:
merging with #1148.

commit 02c1407f (2025-03-16): Add local IP address to About screen: Resolves #622, #1202, #1320, #4128 (#4633)
closed 2020-11-02 by UnchartedBull