UnchartedBull/OctoDash #623: Unable to start OctoDash Elecrow 3.5LCD & RPi3b

Ga1j1n, 2020-05-07:
Following a support request created under #558 [https://github.com/UnchartedBull/OctoDash/issues/558#issuecomment-624634369](url)

I have taken on the feedback and tried other approaches including installing desktop and allowing anyuser to login

As part of this I have an HDMI monitor plugged in and I can see that OctoDash has started and is displaying there but the 3.5" LCD is showing the console, I guess my issue is redirecting the output from HDMI to LCD, Any ideas

Scripts executed to date below

**Install Octopi
Flash image to your storage media using "balenaEtcher".
Using notepad++ edit the "octopi-wpa-supplicant" file for your wifi settings.

Install storage media and boot up Raspberry Pi
Get IP address of Raspberry Pi and connect to it using "Putty"

**Install OctoDash

***Dependencies install:

Run:

sudo apt update

**Install Touchscreen driver:

Run:

git clone https://github.com/Elecrow-keen/Elecrow-LCD35.git

cd Elecrow-LCD35

sudo ./Elecrow-LCD35

reboot

cd Elecrow-LCD35

sudo ./Elecrow-LCD35 270 (If upside down)

reboot

***OctoDash install:

wget -qO- https://github.com/UnchartedBull/OctoDash/raw/master/scripts/install.sh | bash -s -- --ptg

*** Workaround to make 3.5LCD's function

sudo /home/pi/scripts/install-desktop

whne prompted chose desktop autologin = No

sudo reboot

**enable Console Autologin**

sudo raspi-config

** reboot

>Boot Options>Desktop / CLI>Console Autologin

sudo dpkg-reconfigure x11-common

sudo nano /etc/X11/Xwrapper.config

add line:

allowed_users=anybody

Upon reboot touchscreen should be up and running.

UnchartedBull, 2020-05-07:
Do you have HDMI Hotplug enabled? Maybe try turning that off? https://raspberrypi.stackexchange.com/a/2171

Ga1j1n, 2020-05-07:
It seems not, hdmi_force_hotplug=1 is commented in /boot/config.txt

UnchartedBull, 2020-05-07:
I haven’t had a problem like this, so I don‘t know either how to proceed here. But since OctoDash is working fine via HDMI I think this is an issue with your driver / setup. Maybe try googling if there are similar issues (with other apps) reported and how people fixed them. Then try those out and report back which one worked for you. You may brake your Linux (which is half of the fun), so make sure you backup any important data.

closed 2020-05-27 by UnchartedBull

italocjs, 2020-06-06:
Thanks, this helped me too (installing install-desktop and setting Console Autologin),  i'll try writing a guide to help people with this, i was totally lost too, i didnt even remembered that i would have to install the display haha.

UnchartedBull, 2020-06-06:
Please do so and post it somewhere here for everyone to read :)

er9xuser, 2021-01-31:
> 
> 
> Thanks, this helped me too (installing install-desktop and setting Console Autologin), i'll try writing a guide to help people with this, i was totally lost too, i didnt even remembered that i would have to install the display haha.

Did you manage to wrote a guide maybe? My problem is that I have working octodash only over HDMI. Before it was running normaly from touchscreen..but i have to wipe sd card and start from the beginning.