UnchartedBull/OctoDash #624: Possible to add a CPU load and temp display option?

spiff72, 2020-05-07:
**Is your feature request related to a problem? Please describe.**
Sort of.  I am just interested in seeing some "live" stats for the Pi's CPU (load/temp) as a result of the issues around the DLP plugin.

**Describe the solution you'd like**
I think it would be useful to see the Pi's CPU load and temp on the Octodash screen

Something similar to the "[Dashboard](https://github.com/StefanCohen/OctoPrint-Dashboard)" plugin - where they show the CPU load and temperature (but this one

Protoncek, 2020-05-09:
Personally i don't think how this info would be usefull. All Pi's except Zero are "good enough" for octoprint pusrpose, so additional info would just add clutter to the screen and make it more unreadable.

UnchartedBull, 2020-05-11:
I think I'm with @Protoncek here. This would be nice for now sure, it would take quite a big chunk of time to do this with a graph library and stuff. After the DLP bug is fixed, probably no one will use this anymore. Is there another reason to have this available?

spiff72, 2020-05-11:
Probably not - more of a "nice to have".  I wasn't really expecting a graph - just a text field showing the CPU load as a percentage - and maybe the CPU temp.

I-like-stuff, 2020-05-23:
I would like to have the CPU temp, as the raspberry pi 4 can get a bit toasty and having it on the screen would be more convenient. You could implement it using this plug in which shows the temperature on the octoprint web page:
https://github.com/imrahil/OctoPrint-NavbarTemp

UnchartedBull, 2020-05-23:
The plugin you linked unfortunately doesn't have an API, so I can't really use it.

I'll have a second look at this once I have more time again and then decide, whether this is something for OctoDash.

closed 2020-06-22 by stale[bot]