UnchartedBull/OctoDash #631: Add custom actions on "sleep" screen.

savenko-egor, 2020-05-08:
**Is your feature request related to a problem? Please describe.**
I added custom action for enable PSU. If PSU off Octprint can't connect to printer and OctoDash change screen on "sleep". And I can't switch power on PSU from OctoDash.

**Describe the solution you'd like**
Add possibility for adding custom action on "sleep" screen. Or add possibility on\off "sleep" mode.

Thanks!

Protoncek, 2020-05-09:
True, if printer is not connected to Octoprint you can't to anything on LCD, not even shut down printer with custom action.

UnchartedBull, 2020-05-11:
You actually can switch the PSU form the printer back on, IF OctoPrint is still running. Please see this config parameter: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#psu-control-plugin-turn-on-psu-when-exiting-sleep. Is this what you're looking for? There are also some improvements to this deployed with the next version.

savenko-egor, 2020-05-11:
I don't understand how I didn't notice this before. Thank you. I will try use it.

closed 2020-05-11 by savenko-egor