UnchartedBull/OctoDash #640: Current feedrate not correctly shown in the adjust menu

thegarbz, 2020-05-14:
While printing when using the adjustment the current feedrate is not correctly shown in adjustment menu. I'm not sure if this is an OctoDash bug or a lack of capability in OctoPrint.

To reproduce:
1. Start print.
2. Touch screen and click on adjust. 
3. Change feedrate to 75% and change bed temperature from 60 to 65C
4. Correct expected behaviour observed on printer, printer slows down and temp increases. 
5. Touch screen and click on adjust. 

- Expected behaviour: Feedrate shows 75% and Bed temperature shows 65C
- Actual behaviour: Feedrate shows 100%, and Bed temperature correctly shows 65C.

6. Open Octoprint and click on control
- Expected behaviour: Feedrate in Octoprint's control menu shows 75%
- Actual behaviour: Feedrate in Octoprint's control menu unchanged. 

7. Change feedrate in Octoprint's control menu to 80%
8. Correct expected behaviour observed on printer, printer speeds to 80%.
9. Touch screen and click on adjust.
- Expected behaviour: Feedrate shows 80%
- Actual behaviour: Feedrate shows 100%

At present there doesn't seem to be any coupling between the feedrate in OctoDash and the feedrate in Octoprint, however both correctly send the M203 command to the printer. 

If Octoprint does not provide access to the current feedrate value in their API it may be worth at least tracking what it was last set to on the Octodash display.

UnchartedBull, 2020-05-15:
OctoPrint doesn't report feedrate and flowrate via the API. I'll switch some bits around, so at least OctoDash does remember what has been set through the OctoDash UI.

thegarbz, 2020-05-15:
@UnchartedBull That memory should be sufficient. The link to OctoPrint is less relevant. However I'm going to raise a feature request on Octoprint as well so maybe this can be addressed in the future.

Cheers,

PR #643 (2020-05-15): Remember Feed- and Flow-Rate
Fixes #640
closed 2020-05-15 by UnchartedBull