UnchartedBull/OctoDash #647: Filament change to support M600 command rather than self feed

thegarbz, 2020-05-15:
I started writing a bug report against the new v1.5.0 but instead realised this is more of a feature request. 

The addition of filament manager is great, however it attempts to double up a feature that is already automatic in some printers. Specifically it attempts to handle retraction of old filament and purging of new filament. 

For printers with a filament autoload functionality this results in the following problem:
1. Select new filament
2. Hot end heats up.
3. Filament is inserted into the the extruder.
4. Printer begins autoloading filament and purges old filament from the hot end. 
5. a) expected behaviour: Nothing, loading process complete.
6. b) actual behaviour: a second amount of filament is fed and purged through the hot end by Octodash which doesn't realise the printer has done this automatically as part of its autoloading routine. This wastes filament for no reason.

**Workaround / Feature request**
It would be great if we had the option to rather than have OctoDash attempt to change the filament, instead to do the following:
1. Select new filament
2. Hot end heats up.
3. "M600" command is issued to printer. 
And that is it. From there the printer can handle retracting of old filament, loading of new filament and confirmation that enough has been fed to purge the hot end. 

Sidenote: The prusaslicer images and the new filament manager are amazing. Thankyou so much for your amazing work on this to date.

UnchartedBull, 2020-05-18:
Ok that should be doable fairly easy. Do prevent OctoDash from pushing through additional filament just set your feedLength and purgeDistance to 0. This way you will go from the heatup screen directly to the confirm screen and then to the purge screen. By default 0mm will be extruded, but you do have the option to extrude more filament in 10mm steps, if you need to. Then just hit done and you're good to go. This should work just fine with the Prusa printers, right?

I will then just add, that you could configure OctoPrint to send the M600 command after heating up.

thegarbz, 2020-05-18:
Indeed setting it to 0mm prevents the wastage and that is what I have done. 

If we had a tickbox in the settings to send M600 right after heating, then in conjunction with both settings being 0mm it works perfectly. This is actually what I'm doing right now as a workaround. Using your fantastic new filament interface and then when it gets to temperature either using the console to send M600, or going through the printer menu to initiate filament change.

UnchartedBull, 2020-05-18:
I created a version with M600 enabled (can be set via the settings menu). Could you please test that? Also you don't need to put your feedLength to 0 anymore, once M600 is activated the feedLength and purgeDistance will always be 0.

Here is the link: https://drive.google.com/open?id=1LNnV6Vd58XNKxKpidzSetb3CfRTN5KUN

Thanks!

thegarbz, 2020-05-18:
Brilliant, I'll put it through it's paces and report back.

thegarbz, 2020-05-18:
Thank you, this worked so much better than you could possibly know, because I actually had a filament jam, but because it was in the middle of an M600 routine the printer gave me the option of disengaging the extruder and manually pulling the filament out.  If it were in the middle of your manual routine I would have had problems removing the filament 😄 

Sidenote, maybe because of this dev release, I got an error during install, but it hasn't affected performance:
pi@octopi:~ $ sudo dpkg -i octodash_1.5.0_m600_armv7l.deb
[sudo] password for pi:
(Reading database ... 65442 files and directories currently installed.)
Preparing to unpack octodash_1.5.0_m600_armv7l.deb ...
Unpacking octodash (1.5.0) over (1.5.0) ...
Setting up octodash (1.5.0) ...
**_/var/lib/dpkg/info/octodash.postinst: line 10: update-desktop-database: command not found_**
Processing triggers for hicolor-icon-theme (0.17-2) ...
Processing triggers for mime-support (3.62) ...
pi@octopi:~ $

Thanks again for your amazing work.

closed 2020-05-18 by UnchartedBull

LingChristian, 2020-08-26:
Thanks for the nice work UnchartedBull.

After some time with Octodash, I´m wondering, if there is a possibility to just select a filament, instead of load/unload.
I like the fact, that you can detect which spool is currently used and how much is left, but I´m no fan of the load/unload procedure. :/

UnchartedBull, 2020-09-02:
You can either get around the load / unload procedure by setting the feed distance to 0, thus omitting the unloading / loading procedure, or you can replace the default procedure by M600, just turn that on in the config and once the nozzle is up to temperature OctoDash will send the M600 command to the printer and let Marlin handle the rest.

PR #654 (2020-05-18): Support m600
Closes #647