UnchartedBull/OctoDash #648: Support Question: What are the ideal thumbnail dimensions?

thegarbz, 2020-05-16:
Is there an ideal set of dimensions to generate thumbnails for use in OctoDash?
When setting up PrusaSlicer I have adopted the Prusa Mini settings of 16x16 and 220x124. In OctoPrint it simply crops its display. In OctoDash it doesn't.

The result is: https://imgur.com/a/OjnRVpl
Fine in the file selection, but absolutely pointless on the print progress display (7" 1024x600 display)

Does OctoDash benefit from a specific dimension of thumbnail?

EPDCenter, 2020-05-16:
To me, it only appears from the file browser, but when printing, the thumbnail does not appear, I am using PrusaSlicer

UnchartedBull, 2020-05-16:
Any Square aspect ratio should work fine. I think I did 256x256 and don’t have this problem.

@EPDCenter just tap the percentage and will switch around :)

closed 2020-05-16 by UnchartedBull

thegarbz, 2020-05-16:
Thanks, yes square ratios work a bit better. Possibly worth adding to the "Tips and Tricks" section.

EPDCenter, 2020-05-16:
[image]
256x256 thumbnail :S any idea?

thegarbz, 2020-05-16:
That circle doesn't look round, is there something janky with the resolution on your screen which is affecting the layout on the UI?

EPDCenter, 2020-05-16:
Solved, many thanks !!!

UnchartedBull, 2020-05-17:
Has been updated, thanks :)

Dryphter, 2020-07-09:
> Solved, many thanks !!!

@EPDCenter What was the fix for this?   I have OctoDash running on a Pi 3, Adafruit 3.5" PiTFT display (480x320) and have the same issue that you posted above?  The circle on mine is round, but the thumbnail looks like the image you posted and there is a gap between the 'Layer x of x' line and the name of the printer.