UnchartedBull/OctoDash #649: Can't get past "Hey There" screen after fresh install.

Hawkins00000, 2020-05-17:
**What doesn't work?**
after installed and rebooted, I updated the auto-logon, and now I am at the "Hey There" new setup screen, and I don't see mouse and no keyboard input. 

**What did you already try?**
Changed mouse and KB's.
Did and lsusb, and both USB devices are recognized.

General Information:**
 - Hardware [e.g. Raspberry Pi, if you have layout issues please also include your screen resolution]
 - OS Info: OctoPi]
 - OctoDash Version: Latest version

Simply: How do I move past this screen?

Simple question, but nothing I do gets me past this screen,...?

Please and Thank you!
Chris
Hawkins0000@gmail.com

UnchartedBull, 2020-05-17:
You don't use a touchscreen monitor, do you?

OctoDash hides the mouse by default, since it just looks weird, if you're using a touchscreen. You basically have two options now:

Create the config manually: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#creating-the-config-manually
Or slightly adjust your `/.bashrc` file to not hide your mouse pointer. Just change the line `xinit -- -nocursor` to just `xinit`. This should enable your cursor again.

closed 2020-05-23 by UnchartedBull

theshortest, 2020-08-08:
Hey, total linux novice. I love the idea of Octodash but do not have a touch screen yet. I am trying to follow your steps for un-hiding the mouse but I have no clue how to edit the file you reference let alone even find it. Much appreciated!!

theshortest, 2020-08-08:
Found my solution. Issue #717 gave me enough info to run a sudo nano bashed command and got it working

UnchartedBull, 2020-08-10:
Ok that is fairly simple:

- Usually most of your files are living in your user directory (abbreviated as "~", full path will be something like "/home/pi/" - both of these are pointing to the exact same location)
- files that start with a `.` are hidden files by default, so you can't list them with `ls`, if you want to see all files (including the hidden ones) use `ls -a`

Now to your question:
- You want to edit that file with a text editor (nano is the most convenient one, if you're used to GUI programs) so to edit the file execute:
`nano ~/.bashrc`
- This will open up the file and you can make changes (there is no cursor support so use the arrow keys to navigate to the correct place and delete the `-- -nocursor` part)
- Hit CTRL + X to exit
- Confirm with enter to overwrite
- Reboot

theshortest, 2020-08-11:
Hey thanks a lot for the thorough response. I figured it out by following up on another issue I found. The nano ~/.bashrc worked.
I appreciate the description of how the file structure works a little. I have it up and running and, though its not made for a mouse, it works fine. I will get a touch screen in the near future. I've seen it somewhere before but I cant find the list of verified screens, 
While I'm at it, is it possible to extrude filament besides a filament change?

UnchartedBull, 2020-08-18:
Screenlist can be found here: https://github.com/UnchartedBull/OctoDash/wiki/Supported-Devices#recommended-screens. The first three are known to be working great, the bottom 2 should work fine, but I haven't heard anybody specifically mentioning that they work OOB.

You can change your custom actions to include extrude / retract commands. There is currently no other way though.

slicerFPV, 2021-01-27:
I try to get into the .bashrc but when I do I don’t see anything that says Xinit no cursor