UnchartedBull/OctoDash #650: Waking screen from sleep should not trigger touch event on OctoDash

ChaosBlades, 2020-05-17:
**Describe the bug**
When the touchscreen is asleep (backlight is off) waking the screen by touching it also triggers a touch event on OctoDash. When printing for example this brings up the cancel screen and you need to close it every time you wake the screen while printing.

**Expected behavior**
Touching the screen when it is asleep should only turn the screen on. Not trigger a touch event in OctoDash. This would require 2 taps to turn on the printer instead of 1 when using PSU Control or maybe that would be the one exception if possible. This is how all consumer touch screen devices wakes work that I have seen, or they require a double-tap to wake the screen.

**General Information:**
 - Hardware [Raspberry Pi 3B+ and Adafruit PiTFT Plus 3.5"]
 - OS Info [OctoPi 0.17.0]
 - OctoDash Version [1.5]
 - OctoPrint Version [1.4]

Usability Feature Request or Bug? Change if necessary.

UnchartedBull, 2020-05-18:
The problem here is, that this should be handled by Raspbian and not by OctoDash. The clicks get passed straight through to OctoDash and in the app there is no way for me to tell if the screensaver is enabled or not, at least not in a feasible way ... I've yet to find a method on how to tell Raspbian to ignore the wakeup tap ...

Crouze, 2020-05-19:
AFAIK, Xorg cannot support such. Most developers will disable xorg screen blanking and build it into their own application instead. For instance: A timer is run which is reset every time there is a control input; when the timer does trigger the application sets the screen backlight off or screen brightness to 0; if the screen is touched there is a check if the screen was previously set to off: if so then set the screen brightness to the previous value and do nothing else, if not then execute the pushed command. Screen backlight on PI's can be easily controlled by changing /sys/class/backlight/rpi_backlight/bl_power (0 = on, 1 = off, surprisingly), brightness by changing /sys/class/backlight/rpi_backlight/brightness. Haven't tested it myself but this would give much more control than by relying on xorg.

UnchartedBull, 2020-05-22:
Ah ok, then this is kind of linked to #410. I guess I'll track this over there then.

Once I have time to work on this, I'll make sure to let that replace the Xorg native screensaver. Closing this, since it isn't really a bug, it will be fixed with the issue linked above though - if that makes sense.

closed 2020-05-22 by UnchartedBull