UnchartedBull/OctoDash #651: Cancelling a print from the Printers LCD does not cancel the print

ChaosBlades, 2020-05-17:
**Describe the bug**
When the printer is printing and you cancel it by using the printers LCD instead of OctoDash the print head will move to the park position then it will continue to print from the parked positions Z height instead of actually cancelling the print.

**Expected behavior**
OctoDash acknowledges prints cancelled from printers LCD and stops the print.
In addition.
When cancelling a print from OctoDash the printer just stops. It does not move to the park position. This makes it difficult to clear the bed of plastic for the next print.

**General Information:**
- Hardware [Ender 3 Pro, Raspberry Pi 3B+, and Adafruit PiTFT Plus 3.5"]
- OS Info [OctoPi 0.17.0]
- OctoDash Version [1.5]
- OctoPrint Version [1.4]
- Marlin Version [TH3D-Unified-U2.R1.B1]

tiwo85, 2020-05-17:
You have to define a stop Script in Octoprint.

UnchartedBull, 2020-05-17:
This is related to OctoPrint, rather than to OctoDash. OctoPrint (and then automatically OctoDash as well) will support cancel from the LCD, you just need to modify your Marlin firmware to let OctoPrint know, that the print has be cancelled. See here for more information: https://community.octoprint.org/t/how-to-stop-octoprint-from-lcd/7586/3

Regarding the parking of the printhead, this should be handled by scripts from OctoPrint as @tiwo85  mentioned. More information here: https://docs.octoprint.org/en/master/features/gcode_scripts.html#sec-features-gcode-scripts-bundled and here: https://forum.prusaprinters.org/forum/original-prusa-i3-mk3s-mk3-software-archive/octoprint-after-print-job-is-cancelled-gcode-script/.

closed 2020-05-17 by UnchartedBull