UnchartedBull/OctoDash #652: Possible issue when updating?

ruralaccent, 2020-05-17:
I just ran the update script via Putty and received the following warning:

[image]

The update seems to have worked ok and Octodash seems to be running ok, but as I've not run the update script before, I don't know if this is to be expected?

I am running a Pi4 (4GB) with the latest Octoprint version.

UnchartedBull, 2020-05-18:
From what I can see the script somehow didn't get the current release properly. Maybe try again to see if this works now, otherwise just do:

`wget -O octodash.deb https://github.com/UnchartedBull/OctoDash/releases/download/v1.5.0/octodash_1.5.0_armv7l.deb -q` and then `sudo dpkg -i octodash.deb`. After that you can delete octodash.deb with `rm octodash.deb`

ruralaccent, 2020-05-18:
Sorted. thank you!

closed 2020-05-18 by ruralaccent