UnchartedBull/OctoDash #668: Enclosure plugin / Control screen when printer Off

graphiste, 2020-05-21:
Hello, I just discovered Octodash and installed it successfuly. Sorry if my questions have been adressed already, but I wasn't able to find a related topic.

I use heated enclosure controlled by the enclosure plugin. The enclosure plugin also turns On and Off the printer and I have some aditional outputs to control LED lights.

So my questions are:
1. How do I set-up the Enclosure plugin in Octodash. I see some field to enter a sensor data, but dont understand what it is for. Ideally I would like to have access from the screen to the switches I configured and to set the enclosure temperature, like I do in OctoPrint.

2. Octodash displays an error message if the printer is not ON and not connected. Is it possible to have the available controls from the Enclosure Plugin on the screen when the printer is OFF, so I can turn On the printer or set the enclosure temperature from Octodash? Otherwise I still need to do it from Octoprint through my PC.

savenko-egor, 2020-05-22:
Hello!
Read #631 issue.

UnchartedBull, 2020-05-22:
The Enclosure plugin integration currently only allows to read a temperature sensor. So you can enter the ID of any sensor, that is registered within the enclosure plugin and that temperature will then be shown at the bottom of the screen. There is currently no way to control anything within the Enclosure Plugin from OctoDash.

You can turn on and off your printer via OctoDash. For more details about that the issue @savenko-egor linked above and https://github.com/UnchartedBull/OctoDash/wiki/Custom-Actions. But what error messages are you getting? There shouldn't be any error messages, but rather there should be a standby screen.

savenko-egor, 2020-05-22:
I think @graphiste speak about this error: 
[image]

UnchartedBull, 2020-05-23:
That error should not show up on the sleeping screen .. Do you get this every time?

savenko-egor, 2020-05-23:
I got this error only once after turning it on. After I click on message, I see this message:
[image]
All works normal.

UnchartedBull, 2020-05-23:
Normally this error shouldn't show up, as the sleep screen disables the notification service. Did you start your pi with your printer disconnected / off?

savenko-egor, 2020-05-23:
With disconnected.

graphiste, 2020-05-26:
> The Enclosure plugin integration currently only allows to read a temperature sensor. So you can enter the ID of any sensor, that is registered within the enclosure plugin and that temperature will [...]

Thanks, this answered my question. I understand so far Octodash can not interact with the enclosure plugin, just read the temperature. 
You are suggesting to use PSU Control Plugin? I dont think I need another plugin to control RPI's IOs. I am already turning On/Off the printer and the enclosure heater from the enclosure plugin.

The solution for me seems to have 2 physical buttons to RPi IOs next to the octodash screen - one button will turn the printer on locally, the second one will set the enclosure temperature to 50 C.  Now I just have to find how to use pulse signal to turn On the printer from a pulse button, so the enclosure plugin can switch the printer Off automatically after the print job is done.

graphiste, 2020-05-26:
Ooo, I see I can open a web page in iFrame. So maybe I can just open the octopi on the enclosure plugin page and turn On the printer/set the temperature from there?
But how to get to Octodash screen when the printer is Off?
I am getting the error @savenko-egor mentioned above that doesnt let me see the Octodash menu when the printer is Off.

UnchartedBull, 2020-05-26:
There is currently no way to see the OctoDash Main Menu, if the printer is off. Since most of the values wouldn't be available anyways and it would be a lot of work to catch all the errors. I guess you basically have two (or three) options now:

- Switch your PSU over to PSU Control Plugin, which already is supported
- Wait for someone (or me) to add support for PSUs in Enclosure Plugin (or make a PR yourself! :))
- Do that thing with the 2 physical buttons

There probably never will be a way to display the main screen, if OctoPrint isn't connected to the printer ...

closed 2020-05-26 by graphiste

graphiste, 2020-06-02:
I switched to Klipper and noticed Octodash works fine when the printer is Off. I can now open a web page from a custom button [!WEB]http://localhost:5000/#tab_plugin_enclosure and have access to all configured IOs