UnchartedBull/OctoDash #669: Touch not at right point when pressing an icon on screen

jompa68, 2020-05-21:
**What doesn't work?**
When i click on some of the icons on main screen i must move away from the actual icon, a bit down and a bit to the side before it responds to right icon. A bit frustrated to "search" for the the right place to touch.

**What did you already try?**
All is working ok with OctoDash just the touch is not working correctly. I have calibrate and all is ok there but not i OctoDash

**General Information:**
 - rPi 4 with OctoPrint  0.17.0
 - PiTFT Plus 480x320 3.5 TFT+Touchscreen

UnchartedBull, 2020-05-22:
I haven't heard of anyone else having this problem. Could you please confirm that you've setup the screen resolution correctly and maybe retry the calibration? Thanks!

jompa68, 2020-05-25:
Did rerun the script again to re-calibrate the screen and did only run auto calibrate setup and not the manual calibrate and now all is ok. 
Great plugin and now time to play with it :)

closed 2020-05-25 by jompa68