UnchartedBull/OctoDash #675: Possible Issue with Scrolling

ruralaccent, 2020-05-22:
Pi4/4GB
Hyperpixel 4 Screen
Octodash v1.5.0

I seem to have an issue on my two seperate Pi4 install, whereby the bottom item of the list of files does not appear on the screen, other than a small part of the top of the name:

[image]

You can't really read the dir/filename, but you can tap it to select it.

I don't seem to have a scrolling issue with anything else on this screen, so I wonder if it may be Octodash. Does anyone else have this issue?

Many thanks :)

UnchartedBull, 2020-05-23:
Does this only happen on the file screen or are there other things cut off as well, i.e. in the settings or the main screen?

And can you please confirm, that you've setup the resolution of your screen correctly?

ruralaccent, 2020-05-23:
I've got a couple of prints running just now, so will double check those points later today. However, I'm pretty sure that the resolution is fine. Will report back...

ruralaccent, 2020-05-25:
I checked the pi last night and I can confirm that the resolution is ok and the bottom entry is mainly off screen on all of the subdirs. I will try adding more dirs/files to the main menu to see if that is any different.

UnchartedBull, 2020-05-26:
Ok ... Could you please check if the mainscreen (https://github.com/UnchartedBull/OctoDash/blob/master/screenshots/main-screen.png) and settings screen (https://github.com/UnchartedBull/OctoDash/blob/master/screenshots/settings.png) are cropped as well? Maybe also if the file details are fully visible?

The Hyperpixel 4.0 uses 800x480 resolution, doesn't it?

ruralaccent, 2020-05-26:
I can't check the screens just now as both printers are running jobs, but looking at the image on my other open issue, it looks like the top of the screen is not cropped, but the bottom is slightly cropped on the main screen. I hadn't noticed that before.

[image]

UnchartedBull, 2020-05-27:
At least that image doesn't look cropped to me, this looks like the normal padding for 800x480 screen. I'll try to replicate this. It isn't happening on your other machine though?

ruralaccent, 2020-05-28:
It occurs on both of my installations. Both use the same type of Hyperpixel screen.

UnchartedBull, 2020-07-20:
Hi, sorry for my late response to this. I think I fixed the bug, please try the following build and report back, whether that fixed it: https://drive.google.com/file/d/13V0M-ddeYHP-DSQ1b4Da1Gv0XjXUT2kS/view?usp=sharing. Thanks for your patience!

ruralaccent, 2020-07-20:
No worries on the delay. How do I update to that build?

UnchartedBull, 2020-07-20:
Download that file to your raspberry (i.e. with wget) and then just run `sudo dpkg -i <downloaded-file>` and then reboot

closed 2020-07-28 by UnchartedBull

ruralaccent, 2020-08-05:
Sorry I didn't manage to test this. My wife went into labour! I'll update to the new version of Octodash and let you know if all is ok.

ruralaccent, 2020-08-05:
Yes. The new version fixes the issue. Thank you!

UnchartedBull, 2020-08-05:
That's definitely more important :) good luck to all of you and not so many sleepless nights ;)

PR #833 (2020-07-28): prevent breadcrumbs in multiple lines
Closes #675.