UnchartedBull/OctoDash #676: New filament manager

I-like-stuff, 2020-05-22:
**Description:**
As Filament Manager has been abandoned (or so it seems) I am suggesting to add support for spook manager which is made by the same person as the display layer progress plugin in the future. I say future as it is not actually ready for mainstream use yet however in a recent issue on DLP the creator said that they are going to spend more time on this plugin. By the look of the screenshots this plugin can contain much more spool information.

Link: https://github.com/OllisGit/OctoPrint-SpoolManager

UnchartedBull, 2020-05-23:
The plugin by Olli seems to be very new and there was no activity since Jan 11. Let's see how this turns out and wait for some reports. There is also the possibility for someone else to pick up Filament Manager. 

If people who are preferring / using OctoPrint SpoolManager over FilamentMager can give a thumbs-up, that would be a good indication on if and when to consider the change.

I-like-stuff, 2020-05-23:
Yeah, I know there hasn’t been much activity lately but as I said Olli said in a recent discussion that he would work on more on this and his print history plugin that the display layer progress plugin. So far the print history plugin has had a lot of work done to it and i think it is close to release so hopefully work will begin on the spool manager plugin soon

ruralaccent, 2020-05-23:
Give how many folk use the existing plugin, and the hours of effort that will have gone into loading brands and filament spool data, it would be good to keep functionality for it, so long as it still works and is used. For me, Olli's plugin, whilst great, is over-engineered and I wouldn't use half of what it does, so I wouldn't be interested in switching unless the current one breaks down due to lack of maintenance.

UnchartedBull, 2020-05-23:
Let's just wait how this turns out. I want this plugin officially released for at least 1 or 2 months, before considering to implement this and see what people think of this. Keeping this issue open until then.

Kawuezel, 2020-06-29:
12 days ago, someone posted that he would take over Filament Manager. Maybe it's still not dead.

ruedli, 2020-07-12:
Is it a problem, when I use filament on more Thea one printer, each with their own octoprint. Or is the database shared?

mikulik86, 2020-08-24:
Version V1.0.0 of OctoPrint-SpoolManager is out and landed in repository. Seems like a great improvement over Filament Manager.

mikulik86, 2020-08-24:
> Is it a problem, when I use filament on more than one printer, each with their own octoprint. Or is the database shared?

OctoPrint-SpoolManager plans a use of an External Database for the furure.

UnchartedBull, 2020-08-24:
Seems like a good plugin (although there is too much going on for my liking). I guess once 1.1 or 1.2 is released I'm having a more thorough look (DLP created lots of problems) and see if I can integrate this side by side with the OG FilamentManager.

Frix-x, 2020-10-22:
Hello all,
I join this discussion as I switched myself to the plugin "Spool Manager" that I find more complete and clean than the old FilamentManager. In fact, Today on the v1.2 (realeased on September) all the base features of FilamentManager used in Octodash are there I think.
Being able to choose either FilamentManager or SpoolManager would be a very good addition.

I add on top of that a feature request for Octodash on the filament screen to be able to know the currently charged filament and the lenght/weight consummed over the total weight of the spool. Maybe it's best to open a new issue for this specificaly ?

Niper8, 2020-12-26:
Having also switched to spool manager, having also used filament manager, spool manager is definitely the better plugin and I would love to see it be supported.

squireblackdens, 2021-01-03:
Also just switched to Spool Manager and would live to see it supported in Octodash - which is brilliant!

Woodpeckercz, 2021-02-01:
I would like to see this implemented as well

Noble-One, 2021-02-20:
I do like Spool Manager much better and would love to see this implemented.

UnchartedBull, 2021-03-02:
All the "pre-work" is done now, so that the plugin that is handling the filament can be swapped out easily. Hopefully will be included with the next release.

JcMinarro, 2021-03-18:
I have been switched to SpoolManager as well and I can't be happier with it.
Would be great to have integration with it on OctoDash

lightmaster, 2021-03-18:
@JcMinarro As of right now, seems you have to have both running. Filament Manager to supply info to other plugins, and SpoolManager cuz it's much nicer and more feature rich, and so you have SpoolManager's database ready for when other plugins add support.

UnchartedBull, 2021-04-19:
Ok quick update on this issue:

The plugin doesn't have a "real" API yet, but rather an internal API. See https://github.com/OllisGit/OctoPrint-SpoolManager/issues/145. Question is: Would you guys rather wait for the REST API, which might be a while or would you prefer a "beta" implementation. The beta implementation probably will probably fail in some cases and i won't be able to provide support for it, but it should work fine for the majority of users.

Please vote with 👍  for beta implementation and 👎  for waiting until the final API is released and a stable integration can be implemented.

ruedli, 2021-04-19:
The effect of having an API is not clear to me. Would this solve two or more Octoprint servers being able to maintain one database with the filament? This is the reality of most owners that have multiple printers running, on the same Pi or in multiple Pi's. You don't use a pool of spools for every printer separately.

OllisGit, 2021-04-19:
Hi @ruedli,

providing a SpoolManager-API has nothing to do with supporting one single database where multiple OP-Instances were connected.
 It is just a stable way to provide spool-data to external applications like OctoDash.

BR
Olli

FreunderNacht, 2021-06-14:
Hi,
I have also switched to the spoolmanger and would also appreciate to have the data displayed in OctoDash.
I would also be willing to test the beta.

notfixingit3, 2021-08-08:
I have switched to spool manager also, this seems to be way better than the old one

UnchartedBull, 2021-08-11:
The next version of OctDash (v2.3.0). Will finally support SpoolManager 🎉 . Thanks to the help from @OllisGit I was able to create a beta implementation with the internal SpoolManager API. All features are fully supported.

Please be advised though that this is a BETA implementation. So stuff might break if you update SpoolManager (feel free to open issues for that of course). I'm personally planning to switch over to SpoolManager as well soon, so possible bugs should be fixed in a timely manner.

closed 2021-08-11 by UnchartedBull

hifihedgehog, 2022-01-07:
Happy New Year! This issue has been closed, but it has been four months and V2.3.0 hasn't been released yet to close this bug.

sym0nd0, 2022-02-16:
@UnchartedBull any ETA for V2.3.0 to hit main and be released?

longwa, 2022-02-16:
I would use the nightly builds, but it seems they are also stale for the last year. I didn't see any instructions on how to build a custom OctoDash either.

UnchartedBull, 2022-02-22:
new release should be done in a couple of hours :)