UnchartedBull/OctoDash #679: Possible to implement "Z-Babystepping" to Adjustment Screen while printing?

seacred, 2020-05-23:
It would be really helpful, to implement Babystepping in the Adjustment screen to fine tune Z-Hight during printing.

UnchartedBull, 2020-05-24:
Will work on this after the v2 release.

mikekscholz, 2020-06-19:
I left my custom control code originally in Issue #55 but here is my updated code that also works with Marlin 2.0.x+
I know its not a direct implementation but hopefully it helps. 

```
controls:
-   children:
    -   children:
        -   additionalClasses: getLiveZ
            command: M290
            confirm: null
            name: Get
        -   additionalClasses: saveLiveZ
            command: M500
            confirm: null
            name: Save
        layout: horizontal
    -   default: 'Current Z Offset: ???'
        regex: Probe Offset ?Z:?([-+]?[0-9.]+)
        template: 'Current Z Offset: {0}mm'
    -   children:
        -   additionalClasses: babystepUp
            command: M290 Z0.02
            confirm: null
            name: Babystep Up
        -   additionalClasses: babystepDown
            command: M290 Z-0.02
            confirm: null
            name: Babystep Down
        -   additionalClasses: microstepUp
            command: M290 Z0.01
            confirm: null
            name: Microstep Up
        -   additionalClasses: microstepDown
            command: M290 Z-0.01
            confirm: null
            name: Microstep Down
        layout: horizontal
    layout: vertical
    name: Live-Z Probe Offset
```

seacred, 2020-06-19:
I got the exact Config implemented in Octoprint, but it would be awesome to adjust it over the touchscreen.

totalitarian, 2020-07-16:
Would love to see this

UnchartedBull, 2020-07-31:
Will be released with the next version. There are also some pictures attached to the PR if you want a sneak peek :)

PR #867 (2020-07-31): Add Z Babystepping
Adds the ability to babystep Z. Currently doesn't retrieve the value from the printer, but relies on history.

[image]

[image]

Closes #679.
closed 2020-07-31 by UnchartedBull