UnchartedBull/OctoDash #681: Your config is Invalid ??

Thull3, 2020-05-24:
I got invalid cconfig after update to latest version 

Tried to uninstall and install again but still the same message 
Im not to good at code so i cant find the config file ?
Dont know how to type in putty to get to file ??
i Have raspberry pi 3b with 7 inch touch display and octoprint on it

UnchartedBull, 2020-05-24:
More info here: https://github.com/UnchartedBull/OctoDash/wiki/Setup-&-Settings#settings-file. You can either delete the file with `rm` and do the setup again or open it with a text editor `nano` for example to fix it.

Thull3, 2020-05-25:
But the thing is I don’t know how to go to the config file .. what command to open it 

Skickat från min iPhone

I-like-stuff, 2020-05-25:
You would use:
 `sudo nano /.config/octodash/config.json ` To edit it
or
 `sudo rm  /.config/octodash/config.json ` to remove it

If nano isn’t installed which it should install it with:
`sudo apt-get install nano`

UnchartedBull, 2020-05-26:
Probably go with removing the config as @I-like-stuff has already posted. Using GNU Text editors takes some getting used to :)

closed 2020-05-26 by UnchartedBull