UnchartedBull/OctoDash #682: 409 Conflict Issue

ruralaccent, 2020-05-25:
**What doesn't work?**
When I boot up my Prusa MK3S and Octopi with Octodash, I get a '409 Conflict' error. It only happens on one of my installations; the other identical setup (Mk3S, octodash etc.) works fine.

9/10 times, if I tap the error message, it just goes away and the system works fine, but every so often, I tap it and it just reappears, again and again, until I reboot the system/restart the printer.

**What did you already try?**
I have check the IP address configuration and there are no other devices on the network with the same IP address (I've given them static IPs). I've checked Raspi config to make sure the host names are different, which they are. I can't think of any other obvious issues and it's not an error I've had on anything else on the network.

**General Information:**
Pi4/5GB
Octopi 0.17.0
Hyperpixel 4" screen
Octodash 1.5.0

**Additional context**
[image]

Any help/ides greatly received and appreciated!

Thank you.

UnchartedBull, 2020-05-26:
It probably isn't caused by other devices on your network, since OctoDash does get a reply. The OctoPrint API docs say, if a 409 is returned for printer status the printer wasn't operational at that time. Could you maybe check what the text in the bottom right corner says, if that message pops up next time?

ruralaccent, 2020-05-26:
Ok, thanks. I'll check next time. Bottom right of the message window or the screen? If the screen, it may be cut off due to the other issue I'm having!

UnchartedBull, 2020-05-27:
Bottom right yes. Where it reads "operational" in your screenshot :)

ruralaccent, 2020-05-29:
Strangely, I've not had this issue repeat for a while. Hopefully whatever has caused it has passed through. I'll keep an eye on it though and if it becomes an issue again, I reopen this request/start a new one. Thanks for your input.

closed 2020-05-29 by ruralaccent